Ultimaker/Cura #3253: Only One Head Allowed in Cura 3.2.0 for Dual-head Ultimaker 3

cjp0008, 2018-02-07:
**Application Version**
3.2.0 BETA.  Doesn't occur with 3.1.0

**Platform**
Win10x64

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Choose a custom material

**Actual Results**
Cura is setup with my ultimaker 3, but it takes away control over the right extruder and only shows one material option. 
[image]

**Expected results**
Should look like Cura 3.1.0:
[image]

**Additional Information**

fieldOfView, 2018-02-07:
Haven't seen that happen before.... Before you do anything, could you go to Help->Show Configuration Folder, and zip up all the contents of the folder so this situation can be debugged?

On the Printers page of the preferences, go to Machine Settings and check if the number of extruders is set to 2. If that does not help, you can probably fix the problem by adding a new Ultimaker 3 printer.

cjp0008, 2018-02-07:
Sure, it’s attached!

The number of extruders already is set to 2:
[cid:image002.jpg@01D3A038.E56CC580]

However, adding a new Ultimaker fixed the issue.  Thank you!!

nallath, 2018-02-08:
Attachments don't work if you reply by email.

ChrisTerBeke, 2018-05-23:
Was beta, no new info, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke