Ultimaker/Cura #3254: Tree supports generate off build platform

retegic, 2018-02-07:
The new tree supports end up generating support structures off of the build platform if the model is close to the edge.

**Application Version**
<3.2>

**Platform**
<Win 10 64x>

**Steps to Reproduce**
<Place an item that requires support close to the end of the bed.>

**Actual Results**
<Tree support structure start building off of the build platform.>

**Expected results**
<Either slicer cannot slice model, or any feasible alternative.>

**Additional Information**
<Side notes: Both normal supports and tree can be turned on at the same time, this really shouldn't be able to happen. Generating tree supports takes significantly more time to slice than normal supports.>

[image]

jackha, 2018-02-08:
Thanks for your feedback. This is currently a known issue, one of the reasons that Tree support is in Experimental. If given enough priority it will end up in 3.3, but we don't know yet.

BagelOrb, 2018-03-06:
JIRA tag: Cura-5037

Vandrasc, 2018-07-09:
Fixed in the next big release.

closed 2018-07-09 by Vandrasc