Ultimaker/Cura #3255: Webcam feeds not correct

jkennard, 2018-02-07:
**Application Version**
3.2 BETA

**Platform**
Win 10

**Steps to Reproduce**
In the Monitor page, when clicking on the webcam feed, it often doesn't show the correct webcam feed. We have 5 machines and it often shows the feed from the last webcam that you clicked on. 

**Additional Information**
Also, there are intermittent connection issues to these machines. It drops the connection quite frequently (every couple of minutes).

ChrisTerBeke, 2018-02-07:
The networking has been completely refactored in master (merged for the 3.3 release) in order to solve a lot of issues we had with networking (including this). I think the best thing to do is either try the master branch manually or wait for the 3.3 (beta) and report back here if it still occurs for you.

Ghostkeeper, 2018-03-02:
I've tried this switching between 4 printers back and forth multiple times but the video feed seems to switch along fine on the current master version. Hopes up for 3.3.

nallath, 2018-03-05:
As for connection drops; If a machine does not respond for 10 seconds (5 subsequent requests), Cura assumes that the printer is no longer connected.

fieldOfView, 2018-03-05:
I am having issues with the camera feed when switching between octoprint instances. @Ghostkeeper, have you tried switching between non-cura-connect printers?

ianpaschal, 2018-03-27:
Should be fixed in 3.3. Can reopen if not but closing for now otherwise if it's working it just gets forgotten about forever.

closed 2018-03-27 by ianpaschal

fieldOfView, 2018-04-20:
>  there seem to be none available in this building 

So how is the LegacyUM3OutputDevice tested then?

Ghostkeeper, 2018-05-08:
> So how is the LegacyUM3OutputDevice tested then?

By swapping it out (or hacking the check for which class it should use). The LegacyUM3OutputDevice class should still work on modern printers.