Ultimaker/Cura #3257: Mac, delete selected Model by just hitting the delete key, does not work.

tmblue, 2018-02-08:
**Application Version**
 3.1 3.2

**Platform**
 MacOS Sierra

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
 Have an item on the build plate, switch windows, click it and hit the delete key, nothing happens, one has to right click and use delete selected model, this is incorrect, I should be able to at anytime hit the delete key to remove a selected model (and sometimes it  works, but most times it requires a right mouse click and delete selected model)

**Actual Results**
 . 95% of the time it won't delete with just pressing the delete key

**Expected results**
 100% of the time if I have a model selected I should be able to hit the delete key and make it go poof, into the ether.,

**Additional Information**

None, noticed it in 3.1 still there in 3.2, has to be more Mac users that this is bothersome? This is my second Mac , this one does it as well, so it's software.

Vandrasc, 2018-02-08:
Hi @tmblue, are you using the Fn+Delete shortcut in order to remove the model and it doesn't work? We had an issue but I thought we fixed it in 3.2.

Ghostkeeper, 2018-03-02:
Didn't we also allow backspace as delete just for this reason?

closed 2018-03-02 by Ghostkeeper

LeviMThomas, 2019-01-05:
The Mac Delete key is a Delete key... It is literally a key with nothing on it at all but the word, "Delete" ! lol.

commit acbc6893 (2018-03-02): Also allow backspace to delete models

See the documentation in the code as to why. Macintosh keyboards...

Fixes #3257.