Ultimaker/Cura #3258: Estimated filament length shown incorrect in 3.2

foxel, 2018-02-08:
Cura 3.2 started showing incorrect filament length for a project. It's about 2.6 times lower than it was shown in Cura 3.1.
[image]

**Application Version**
3.2.0

**Platform**
Linux

**Steps to Reproduce**
Load model, slice

**Actual Results**
Incorrectly estimated filament length

**Expected results**
Correctly estimated filament length

**Additional Information**
Interesting that estimated cost looks like calculated with correct length in mind
https://www.dropbox.com/s/p56p6cohl5zj4wk/K_Headphone_Hanger?dl=0

smartavionics, 2018-02-08:
Similar to https://github.com/Ultimaker/Cura/issues/3117 ?

ChrisTerBeke, 2018-02-08:
Duplicate indeed.

closed 2018-02-08 by ChrisTerBeke

foxel, 2018-02-08:
I don't see it's a duplicate. In my case is not driven by any filament diameter change. Is just constantly 2.6 times smaller than expected

Ghostkeeper, 2018-03-02:
The issue #3117 says it only occurs when changing filament diameter, but it was actually a matter of obtaining the diameter from the wrong location in our data structures and was also occurring without changing it manually.