Ultimaker/Cura #3260: LAYER VIEW Issue in a CAR DESIGN.

soniashish1009, 2018-02-08:
APPLICATION VERSION is **CURA 3.2.0** stable.
 
CURA Platform **(LAYER VIEW)**

I have been using CURA for about 10 months. As a regular work when I have used **_CURA_** to open a file of the CAR designed in the **_SKETCHUP_**. I saw an error. I have seen the SOLID VIEW & LAYER VIEW does not match. 

Actually, I want to hollow the back portion of the CAR which is L- shaped. So in the SOLID VIEW as I am seeing the Design it is same as I want. But in the LAYER VIEW, it is not hollow as I want.

So I want that L- shaped pipe to be hollow in the LAYER VIEW.

I have attached some screenshots, which will give you better knowledge.

[image]

[image]

[image]

[image]
[image]

 Please help ASAP. It's very important to solve this issue.

ChrisTerBeke, 2018-02-08:
It seems your model is not a single solid volume, which makes it hard for Cura (or any other slicer for that matter) to get the intended results

fieldOfView, 2018-02-08:
I agree that there’s an issue with your model, even if it is not directly obvious in solid view. The xray view probably shows some red areas. Sketchup is notorious for producing models that are broken for 3d printing.

You can try un-setting the “Union overlapping volumes” setting. Use the search in the custom sidebar to find this setting.

Ghostkeeper, 2018-03-02:
If you are at liberty to share a model that reproduces this issue, we could tell you more.

ChrisTerBeke, 2018-05-23:
Issue with model, not Cura. Closing.

closed 2018-05-23 by ChrisTerBeke