Ultimaker/Cura #3262: Cura 3.2 crashing on network access

Kernelhacker, 2018-02-08:
Cura 3.2 from Ultimaker's website
Windows 10 x64

Reproduction method 1:
 - Enable "Send (anonymous) print information"
 - Slice/Prpare an object
 - Save to File 
 - Crash
Reproduction method 2:
 - Install  Octoprint plugin
 - Edit printer settings
 - Enter octoprint API key (any keypress will do)
 - Crash

From what I read elsewhere it is probably a problem, because QT is not (yet) included in the recent  3.2 release.

nallath, 2018-02-08:
Could you share logs?

ChrisTerBeke, 2018-02-08:
It's most certainly the Qt issue that will be resolved by upgrading to Qt 5.9+ (which is planned for the next release I just saw).

Kernelhacker, 2018-02-09:
Here are the logs:
[cura.log](https://github.com/Ultimaker/Cura/files/1709969/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/1709970/stderr.log)

Ghostkeeper, 2018-03-02:
Yeah this is a duplicate of https://github.com/Ultimaker/Cura/issues/2622

closed 2018-03-02 by Ghostkeeper