Ultimaker/Cura #3263: 3.2 Adaptive Layers don't work for merged objects as expected

skobkars, 2018-02-08:
If object consists of two (or more?) merged parts then selecting "Use adaptive layers" causes strange issues and artefacts.

**Application Version**
3.2

**Platform**
MacOS

**Steps to Reproduce**
Add two STL files to the project, select both and merge, or change infill for the top part and merge.

**Actual Results**
Slicing either crashes on the first layer, or after completing bottom part of the merged object

**Expected results**
Fully sliced object

**Additional Information**
[image]

ChrisTerBeke, 2018-02-08:
Adaptive layers is experimental and not tested with multiple / merged objects. We'll add this to the list of known issues so we can further improve it in the next iteration.

ChrisTerBeke, 2018-02-08:
Devs: CURA-4933.

closed 2023-06-24 by github-actions[bot]