Ultimaker/Cura #3266: Printer profile "Anycubic i3 Mega" material drop down menu is missing

mukisking, 2018-02-08:
When Printer profile "Anycubic i3 Mega" the material drop down menu is missing!
You dont can selekt own material profiles.
Is also no apparent which material profile is selektet?
If i change on printer "prusa i3" the menu is still there.

**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-02-08:
This was changed here: https://github.com/Ultimaker/Cura/commit/948497c9a88bb23491d449d0c2231c56037da8dc
@thetobby, can you clarify why this was changed?

thetobby, 2018-02-08:
Hi @mukisking  The material drop-down menu is missing since it has been disabled. 
Cura allows developers to choose if they want to add material presets or not for their printers.
I added the printer profile so it was never in Cura enabled before me, just in an early test that got pulled.
There are other printers with the same configurations in Cura.

Since adding a whole new printer with presets and material presets is a very time consuming I decided for this first goal was to only add printer with quality presets. Later if someone wishes to spend the time or I have time they may add the materials as well.

closed 2018-02-08 by mukisking