Ultimaker/Cura #3267: All printers, profiles and materials are missing in Cura

codemonkeh, 2018-02-09:
**Application Version**
3.1.0

**Platform**
Windows 10 Pro x64

**Qt**
n/a

**PyQt**
n/a

**Display Driver**
nVidia GeForce 390.65

**Steps to Reproduce**
1. Open Cura 3.1.0

**Actual Results**
All custom or standard printers, profiles and materials are not loaded. Prompt to add new printer is displayed.

If you go to Preferences > Profiles, a blank dialog is displayed. Clicking on *x* crashes Cura.
[image]

**Expected results**
Cura should operate normally, load configured printer / profile and display the slicing GUI.

**Additional Information**
I recently enabled support for Python development in Visual Studio 2017 and this has installed Python 3.6.3. 
I suspect this may be related, however i likely already had cura open and this did not immediately cause an issue as i was successfully able to print last night. 

I have tried the following with no affect:

* Upgrading to Cura 3.2.0 - (crashes on first load)
* Removing Python support in VS2017
* Reinstalling Cura 3.1.0

Logs and configuration attached, please let me know what additional information can be provided. Thank you.

[cura-log-except.log](https://github.com/Ultimaker/Cura/files/1709259/cura-log-except.log)
[cura-log-except2.log](https://github.com/Ultimaker/Cura/files/1709312/cura-log-except2.log)
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/1709264/cura.cfg.txt)

codemonkeh, 2018-02-10:
Okay I have tested various situations:
* Reinstalling Python 3.6 x64 or x86, or removing one or the other or both
* Reinstalling Cura 3.1
* Deleting the cache or user files

And my printer was not loaded on start-up (with various errors in the log file), this in turn lead to no materials or profiles being loaded. On further inspection In realised that my printer profile had been corrupted and is full of null characters, see attached.
[Creality+CR-10.global.cfg.txt](https://github.com/Ultimaker/Cura/files/1713298/Creality.CR-10.global.cfg.txt)

After i added a removed and added a new printer profile lo and behold all of my custom profiles and materials reappears, what a relief - i have spent many hours tweaking my profiles.

How the profile was corrupted remains a mystery that i cannot attribute fault to. I would suggest that some better validation when loading profiles would be beneficial and am uncertain if this issue should be considered closed?

codemonkeh, 2018-02-10:
I should note that it is not possible to start Cura 3.2 without configuring a printer profile, even if it is subsequently deleted from disk. However if the printer profile is corrupted the application will simply crash when loading.
It does allow you to reset configuration, but it does not validate the configuration and allow you to easily isolate the particular erroneous configuration.

Ghostkeeper, 2018-03-05:
The crash seems to occur because the files are indeed corrupted to null bytes. How that could have happened is a mystery to me. But Cura 3.2 should give you the option to "backup and reset" your user configuration in the crash dialogue that occurs.

Or you could manually go into `C:\Users\<codemonkeh>\AppData\Roaming\cura` and delete the old versions (3.1 and before) and in the 3.2 folder delete everything related to the CR-10 files.

We shouldn't crash on this sort of error, no. We'll have somewhat better validation strategies for 3.3.

closed 2018-03-05 by Ghostkeeper