Ultimaker/Cura #3268: Only Ghost in "layer view"

BlackDogNZ, 2018-02-09:
**Application Version**
<-- 3.2 -->

**Platform**
<-- Mac High Sierra 10:13.4 -->

**Qt**
<-- The version of Qt used (not necessary if you're using the version from Ultimaker's website) -->

**PyQt**
<-- The version of PyQt used (not necessary if you're using the version from Ultimaker's website) -->

**Display Driver**
<--  Intel HD Graphics 6000 1536 mb -->

**Steps to Reproduce**
<-- When viewing imported Jobs in Cura 3.2 in the "Layer View" setting -->

**Actual Results**
<-- What happens after the above steps have been followed (The "Ghost" image of the Part appears but no layers are shown. No Change when using the "Layer Slider") -->

**Expected results**
<-- What should happen after the above steps have been followed (A Layer View of the Part to be printed should Show in it's Print Layers and these should be viewable using the "layer slider") -->

**Additional Information**
<-- Extra information relevant to the issue, like screenshots (replace this text) -->

Vandrasc, 2018-02-09:
Hi @BlackDogNZ, is it oke with you to share the project file with us so we can investigate what the problem could be?

diegopradogesto, 2018-02-09:
The log file is also very helpful.
It is very weird that you can see the "layer slider" but no layers. Can you attach a picture?

xdbgjhjreedwef, 2018-02-11:
I have the same problem with cura 3.2 on gentoo, Qt version 5.9.4
[cura.log](https://github.com/Ultimaker/Cura/files/1714242/cura.log)
[image]

BlackDogNZ, 2018-02-11:
UPDATE:: I can get Layers to show if I click the 'Play" button, then slide through the layers. as soon as the "Play" stops, the layers disappear again. Thus, not able to View layers with out the Model being in "Play" mode.

BlackDogNZ, 2018-02-11:
At @Vandrasc This happens on all files I try and all models.

diegopradogesto, 2018-02-12:
I cannot reproduce it. It is related with the calculation of the layer thickness. Is it happening also if you select "Material color" in the color scheme?

I'm wonder if it can be setting dependant. Can you send me a project file? (File -> Save Project...)

xdbgjhjreedwef, 2018-02-12:
I cannot save project. I get an error "There are not file types available to write with!".

xdbgjhjreedwef, 2018-02-19:
Version 3.2.1 did not fix the issue for me. Layer view still hangs at "processing layers 99%".

xdbgjhjreedwef, 2018-02-19:
I did not have libSavitar, so 3mf plugins were not working. I've installed it now.
Here is the project (I've added ".zip" to filename so github would accept it):
[KSE_cube_50x50x5.zip](https://github.com/Ultimaker/Cura/files/1736153/KSE_cube_50x50x5.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/1736154/cura.log)

xdbgjhjreedwef, 2018-02-28:
Ok, that was my mistake. I had curaengine version 3.0, with version 3.2 problem is resolved for me.

diegopradogesto, 2018-02-28:
@BlackDogNZ can you try with 3.2.1 version that was release a few days ago?

diegopradogesto, 2018-03-16:
I'm closing this ticket since we didn't get any feedback. You can open it if you still see this error.

closed 2018-03-16 by diegopradogesto