Ultimaker/Cura #3269: machine_extruder_start_code at start of print

maukcc, 2018-02-09:
At the very start of the print between the start_gcode and the actual printing, there is no machine_extruder_start_code. 
This is false as it is the start of an extruder. I take it that this is actively taken out for some reason. Can someone explain that reason.
Or put it back in.

fieldOfView, 2018-02-09:
As far as I can remember it was not "taken out", but the extruder start/end gcode are only considered for tool *changes*, I guess because that was the functionality that was needed to support the UM3.

diegopradogesto, 2018-02-09:
The "machine_extruder_start_code" is still there. Can you please give more information about what do you mean? If you fill the form we can, for instance, take a look at the right version of Cura you refer. Also the steps to reproduce it are very helpful.

maukcc, 2018-02-09:
We should consider every start of a print as a toolchange, as we have no idea with which extruder we were printing before. UM3 does not use any machine_extruder_start_code so no wonder that it works there :)
Also, if this is implemented the start gcode can be considerably easier.

@diegopradogesto this is the start of the file (look almost at the bottom):

> ;FLAVOR:Marlin
> ;TIME:177
> ;Filament used: 0.093558m
> ;Layer height: 0.2
> ;Generated with Cura_SteamEngine 3.2.0
> T0
> M104 S190
> M104 T1 S160
> M82 ;absolute extrusion mode
> 
> M92 E159 ;2288 for V5 extruder
> 
> M140 S50
> M104 S120 T1
> M104 S120 T2
> M104 S120 T3
> 
> G21
> G90
> M42 S255 P10 ;chamber lights
> M42 S255 P11 ;fume extraction
> M204 S300 ;default acceleration
> M205 X10 ;default jerk
> 
> M117 Homing Y ......
> G28 Y
> M117 Homing X ......
> G28 X
> M117 Homing Z ......
> G28 Z F100
> G1 Z10 F600
> G1 X70 Y20 F9000;go to wipe point
> 
> M190 S50
> 
> M117 Heating for 50 sec.
> G4 S20
> M117 Heating for 30 sec.
> G4 S20
> M117 Heating for 10 sec.
> M300 S1200 P1000
> G4 S9
> 
> M117 purging nozzle....
> T0
> G92 E0;set E
> G1 E10 F100
> G92 E0
> G1 E-1 F600
> 
> M117 wiping nozzle....
> G1 X1 Y24 F3000
> G1 X70 F9000
> G1 Z10 F900
> 
> M104 S21 T1
> M104 S21 T2
> M104 S21 T3
> 
> M117 Printing .....
> 
> 
> *******************************************************
> *** this part should be machine_extruder_start_code ***
> *******************************************************
> 
> 
> ;LAYER_COUNT:5
> ;LAYER:0
> M107
> G0 F9000 X91.476 Y108.034 Z0.3
> ;TYPE:SKIRT
> G1 F2400 X91.719 Y107.843 E0.01542
> G1 X92.003 Y107.72 E0.03086
> G1 X92.333 Y107.674 E0.04748
> G1 X132.315 Y107.675 E2.04219
> G1 X132.622 Y107.715 E2.05764
> G1 X132.908 Y107.832 E2.07306
> 
> and so on, and so on

Ghostkeeper, 2018-03-05:
It's a question of semantics, because anything in the extruder start g-code can also be put in the global start g-code.

The `machine_extruder_start_code` setting is in fdmextruder.def.json, which when this was put in place was only used for multi-extrusion printers. We considered it only relevant for multi-extrusion printers, so we put it there.

Now that all single-extrusion printers also use `machine_extruder_start_code`, we can consider putting the extruder start code at the beginning as well.

Currently the extruder start and end g-code is only used by Cartesio printers and the Custom FDM Printer.

fieldOfView, 2018-03-05:
> It's a question of semantics, because anything in the extruder start g-code can also be put in the global start g-code.

If eg the second extruder has a different start gecode than the first extruder, then putting the extruder start gcode in the machine start gcode is not a viable option because it would have to depend on the first extruder used.

maukcc, 2018-03-05:
So the point is that when you put the extruder start code in the global start code, you always have to start printing with the same extruder(probably T0). It is not possible to start printing with another extruder(T3) as it will not go through its(T3) extruder start gcode, but rather do the T0 start code that was set in global start.

BagelOrb, 2018-03-06:
> I take it that this is actively taken out for some reason.

The extruder start gcode was indeed actively taken out. As I remember the extruder start gcode was at some point introduced at the start of a print.

I don't really remember why it was taken out, but in the mean time stuff has changed again. Now the first time an extruder is used (which must be on the first layer) the nozzle is primed with a priming gcode; I think G280. The first extruder start is therefore not a normal extruder start. The amount of unretraction depends on how the last print ended, which is information unavailable to CuraEngine.

As the extruder_start_gcode is empty for the UM3 I don't see any reason not to accept a PR which puts the extruder start gcode back in.

However, I'm not in charge of such a PR.

Ghostkeeper, 2018-03-08:
I gave this a quick try. Adding the extruder start g-code at the start is really easy. However adding the extruder end g-code at the end is a bit harder since the piece of code that normally adds the ending g-code is executed after all the per-extruder settings are removed from memory.

But our project manager removed this from our planning now. We won't be getting around to implementing this any time soon.

maukcc, 2018-03-08:
The end gcode is not that important as you will stop everything, and that can be done by the machine_end_code.

UticaTechClub, 2018-12-04:
I was very happy to find this discussion as I am about to give up on Cura just because of this.

We are developing a multi-extruder configuration and every extruder must be picked up at the particular location and returned back to the **same** (!) location.
In a multi extruder case, Cure uses `T0` as the first one. This makes it easier to address the absent invocation of the first tool's `machine_extruder_start_code` through `machine_start_gcode` section.

But the real difficult issue arises at the end, when `machine_extruder_end_code` is not called on the last used extruder. This cannot be addressed using `machine_end_gcode` section because there is no way to know which extruder was the last one and only `machine_extruder_end_code` knows the "parking" location of the corresponding tool. 
I considered python script invocation using `"value" : ""` property, but there is no property which describes the last extruder (or the current extruder).

This post is about the same issue:
[https://github.com/Ultimaker/Cura/issues/4909](url)
and this post to some extent as well:
[https://github.com/Ultimaker/Cura/issues/4923](url)

I humbly suggest that without properly handling of the first `machine_extruder_start_code` and last `machine_extruder_end_code` it would be impossible to use Cura for any advanced multi tool configurations.

With it's simplicity and seamless slicer integration, Cura shines on a single extruder machines.
My prediction is that with the nice new marketplace interface and multi-material support, this issue will be discouraging multi-material developers of true multi-extrusion systems.