Ultimaker/Cura #3271: +/- 0.1mm jog controls not working

The--Captain, 2018-02-09:
**Application Version**
Cura 3.1, 3.2

**Platform**
Ubuntu 14, Linux AppImage (unpacked with bsdtar)

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Add a custom USB-connected printer and select Marlin flavor firmware
Attempt to use the jog controls

**Actual Results**
Jog controls of +/- 10mm and 1mm work as expected along all axes, 
0.1mm jog does not work at all in any axis

**Expected results**
Jog controls should work for all available increments, especially 0.1mm

**Additional Information**
I have sniffed the serial connection, and Cura sends (for example)

G91
G0 X0 Y0 Z10.0 F3000
G90

G91
G0 X0 Y0 Z1.0 F3000
G90

for Z axis jogs of +10mm and +1mm increments respectively, but sends

G91
G0 X0 Y0 Z0 F3000
G90 

for a Z axis jog of +0.1mm (Z0 ?!?!  That does nothing!).   Identical issues are encountered when using negative axis adjustment values.  All results have been explicitly verified using a USB serial sniffer (i.e. USB-capable pcap accessed via tcpdump).

Vandrasc, 2018-02-09:
Ah, yes, we are aware of that, but in the moment of the implementation, the FW did not accept that value. @ChrisTerBeke could you please explain? Thank you!

The--Captain, 2018-02-09:
BTW, the USB sniffer reveals that the Manual Control window button in Slic3r sends the correct value of Z0.1, which the firmware processes correctly, and adjusts the print head accordingly.

fieldOfView, 2018-02-09:
> Ah, yes, we are aware of that

I implemented the jogging, but I was not aware of that. Can you elaborate @Vandrasc ?

The--Captain, 2018-02-09:
I'm not sure where the perception that the FW wouldn't handle the value is originating - my Marlin FW is older than the implementation of the jog function in Cura 3, and it accepts values of 0.1mm for relative movement, as mentioned.

The--Captain, 2018-02-11:
The issue appears to occur at line 24 of plugins/USBPrinting/USBPrinterOutputController.py, such that:

self._output_device.sendCommand("G0 X%s Y%s Z%s F%s" % (x, y, z, speed))

truncates (or perhaps rounds - my testing is not utterly complete) the fractional portion of all variables (i.e. x, y, z, and speed), such that a supplied value of 0.1 winds up as 0.  I have verified this by temporarily replacing "Z%s" with "Z0.1", e.g.:

self._output_device.sendCommand("G0 X%s Y%s Z0.1 F%s" % (x, y, speed))

, which does cause the printhead to move relatively by +0.1mm on the Z-axis whenever *any* jog command is issued, as expected.  

I am not a python coder, so I cannot begin to guess why this is happening;  I can only hope to supply some glimmer of direction to those who can actually fix this.

I must say I *am* surprised that I am apparently the first one to notice this (OK, maybe I'm not and my search skills are just sub-par).  I mean, it does seem like something that might affect others, rather than something localized to my system.

The--Captain, 2018-02-12:
I fixed this in the most stupid way possible, i.e.  multiplying values supplied to moveHead() by 10, and then dividing them back again by 10 inside the moveHead() function.

Stupid, but it will get me by until it's fixed properly by someone who knows how.

Vandrasc, 2018-02-12:
@fieldOfView this is a good reason for you to come back :)  At that time, in the moment of the implementation, the FW would not pick up the 0.1 value for some of the printers (eg UM2 with UM gcode flavor), that is all that the messenger got, I am afraid. The ticket was set on approved and waiting for the user feedback..which we now received. How to handle it further? Any PR coming anytime soon?

Ghostkeeper, 2018-03-06:
It's just a simple matter of the Qt bindings only accepting `long`s, which rounds everything down to integers. Changing those to floats makes it work.

It works a bit unreliable on my printer. Not every command seems to register. Most of them do register though. Maybe this is because it's a UM2...

commit 3e8f29d3 (2018-03-06): Allow floating point values for moving print head

The X, Y, Z coordinates and speed don't necessarily have to be full millimetres or millimetres per minute.

Fixes #3271.
closed 2018-03-06 by Ghostkeeper