Ultimaker/Cura #3272: rotate option

espadan3d, 2018-02-09:
Hi
Thanks a lot for provide Cura.
I use of Cura 2.6(64bit) in Windows 10, when I install 3 version or higher don’t show rotation icon. 
Please help me
Regards
[image]

ianpaschal, 2018-02-09:
Can you post some log information? My guess is that the Rotation Tool, which is a plugin, is not being correctly loaded. This will throw some errors in the log file.

espadan3d, 2018-02-09:
Hi
Thanks 
I check it not active in plugin
I active it and close, open program solved.
In Cura One can print a piece with several layers of thickness(for example 0-5 mm height by 0.3mm and 5-10 mm by 0.3mm layer?
[image]

ianpaschal, 2018-02-09:
Great! Glad that solved it!

> In Cura One can print a piece with several layers of thickness(for example 0-5 mm height by 0.3mm and 5-10 mm by 0.3mm layer?

In general, please open separate tickets for separate questions, but since it's a short one:

Yes. There is an experimental feature called adaptive layer height. If you are using version 3.2, open preferences, go to settings, and scroll all the way down. There you can check the adaptive layer height boxes and you'll see those options in the sidebar.

closed 2018-02-09 by ianpaschal

espadan3d, 2018-02-09:
Thanks a lot