Ultimaker/Cura #3273: Cura 3.2 Heated Bed Issues on Printrbot Play

JAllenWatkins, 2018-02-09:
**Application Version**
3.2

**Platform**
Printrbot Play with a extended and heated bed
**Qt**
Ultimaker website version

**PyQt**
See above

**Display Driver**
Intel HD Graphics 520

**Steps to Reproduce**
Trying to print anything with heated bed set a 60c

**Actual Results**
The bed heats up but the monitor screen shows the bed temp as 0c and the hotend never heats up. Thermistor is fine & is in contact with bed. No breaks in wires from heated bed or from thermistor.

**Expected results**
The hotend should heat up and the print should start.

fieldOfView, 2018-02-09:
Can you try sending the same gcode file with a different print host, such as printrun?

Ghostkeeper, 2018-03-05:
This sort of issue is impossible for us to reproduce or fix if we don't have the Printrbot Play. I'm hoping that someone that does have such a printer can pick it up.

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant