Ultimaker/Cura #3275: Cura 3.2 stops working towards the end of slicing

ABSingh1952, 2018-02-09:
**Application Version**
 cura 3.2 and cura 3.1

**Platform**
 Windows 7 Pro

I am really disappointed with Cura software versions 3.1 and 3.2. I installed 3.2 today, then opened an .stl file. Cura starts to slice the file. Towards the end, a message appears - "cura.exe has stopped working". It never completed slicing. I tried slicing 3-4 other files but same message appears.

Same problem existed with Cura 3.1.

I am currently using Cura 15.04.6 on the same Win 7 computer. It is working without any problem. Can you please check the problem? Thanks.

A.B. Singh

fieldOfView, 2018-02-11:
What is your layer height, and what is your initial layer height?

ABSingh1952, 2018-02-23:
Layer ht. is set to .04mm. There is no place to set Initial Layer height. But I have set Top/Bottom thickness to 0.6mm and wall thickness to 0.4mm.
I don't think these values matter. During last few days, I have tried all available versions after version 15.04.6. Today, I tried latest 3.2.1 version. They all crash towards the end of slicing. I have tried many different files. These are very simple objects. I hope your team can find a solution for this problem. Best! - A.B. Singh

ianpaschal, 2018-02-23:
There is a place to set initial layer height. If you can't see it in the sidebar, open the preferences and then in settings search for it and check the checkbox. It will then appear in the sidebar. You can also check the checkbox for all settings, although the interface may slow down if looking at _all_ settings.

In the mean time though, can you please upload the simple objects that are causing the crash? This way we can test them on various machines and most recent builds to see if the issue is being caused by the objects, the software, or if neither, your machine. From there we can diagnose further.

fieldOfView, 2018-02-23:
There’s also a neat Search box above the custom settings list

rotaryfurball, 2018-03-12:
same problem

Ghostkeeper, 2018-04-11:
I can't reproduce this problem on my end. I recreated the model in question (since the PDF isn't the actual model but just a screenshot of it). Set layer height to 0.04mm and 0.3mm initial layer height, but it just slices fine.

[image]

Maybe it would help to create a project file? That contains the entire set-up including the model and which printer/profiles you're using.
You can rename the file to .zip and Github will accept it.

ABSingh1952, 2018-04-26:
I was waiting for the newer version of Cura - hoping that the problem will be solved. Today, version 3.3.1 was released. So, I uninstalled version 3.2, and installed the latest 3.3.1. Sorry - same thing happend. Towards the end of the slicing, Cura stopped working.
As you have suggested, I have renamed the file from .stl to .zip and attached below. I hope you can reproduce the problem and then fix it.

Best of Luck!
A.B. Singh

[Test_Cylinder_02mm.zip](https://github.com/Ultimaker/Cura/files/1949285/Test_Cylinder_02mm.zip)

ianpaschal, 2018-04-26:
Something went wrong with your project file. The .zip file seems to be corrupted.

fieldOfView, 2018-04-26:
Rename the .zip to .stl (it is a binary stl file)

ianpaschal, 2018-04-26:
Ah I tried .curapackage and .3mf 😆

ianpaschal, 2018-04-26:
This model slices just fine for me. Can't reproduce the issue. @ABSingh1952: Please upload your Cura logs (see: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations). It looks like the logs are the only way we can see the problem in action.

ABSingh1952, 2018-04-26:
Below is the log file attached.

**Info about my setup**
1) I cleared the previous logs and started with a blank log file. Opened Cura 3.3.1 at 15:50 PM. Waited for few minutes then loaded a **.stl** file at 15:52 pm. 
2) At the time of slicing, 3D printer was not connected through USB port. A few minutes ago, I tried with the printer connected, but Cura still crashed.
3) We have **Ultimate MK11** printer by Monoprice. Cura 3.3.1 does not have this in its list, so I created a Custom FDM Printer.
4) We also have Cura 15.04.6 version installed on this Win 7 Pro computer. It is working fine.
Hope this info will help. Thanks.
[cura.log](https://github.com/Ultimaker/Cura/files/1952767/cura.log)

ianpaschal, 2018-04-26:
I have some ideas; I'll discuss with a colleague on Monday (we have a long weekend in the Netherlands this week).

In the mean time, two more things you can do to help are: try using Cura without any anti-virus or firewall running to make sure that's not blocking the communication between Cura and the slicing engine and also save your cura project as a .curaproject and add .zip and upload it. If you do this, we'll have all the settings, your custom machine definition, model, etc.

ABSingh1952, 2018-04-30:
I have tried both suggestions.
1) Antivirus turned OFF. It still crashed towards the end of slicing.
2) Have created a project file and attached below.

Best of luck!
[CFDMP_Test_Cube_1x6mm.curaproject.3mf.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1961231/CFDMP_Test_Cube_1x6mm.curaproject.3mf.curaproject.3mf.zip)

ianpaschal, 2018-05-01:
Thanks for the file. Now I can test with every variable the same as you with the exception of one: our system. It slices perfectly fine when running Cura from source code as well as on 3.3.1 (latest release).

This means, unfortunately, that there's something going on in your system which is blocking it.

Can you try removing all existing Cura installations and using 3.3.1?

ABSingh1952, 2018-05-03:
I followed your instructions - uninstalled both Cura 15.04.6 and 3.3.1; rebooted the PC and then installed Cura 3.3.1. Same problem - stops working towards the end of slicing.

Then I reinstalled Cura 15.04.6 - and it works. I don't think two versions interfere with each other. My guess is that my Windows 7 Pro needs some plugin etc. which is missing. I have down loaded all updates. I don't think it is a hardware issue.
Hope your team can find the problem.

Best!

ianpaschal, 2018-05-29:
Since we haven't been able to reproduce this for upwards of a month, I need to close it. Simply can't fix bugs when they aren't visible.

closed 2018-05-29 by ianpaschal