Ultimaker/Cura #3276: Retraction extra prime amount still has to be positive

HClO4, 2018-02-10:
Sadly the retraction extra prime amount still can't be negative. I need an negative amount to avoid getting blops.

In #2392 it was mentioned that this issue will be fixed in the following version. In Cura 3.2 the issue is still there :(

HClO4, 2018-02-17:
No one who has the same problem?

devils27, 2018-02-26:
i do, they said it was going to be fixed 3 releases ago and still the same thing.  i dont get it.  smartavonics posted a simple fix to literally just allow cura to allow a negative value the the "extra prime amount" spot in the code and still nothing

ianpaschal, 2018-02-26:
Duplicate of #2392 and #2667. We'll continue the conversation in those threads.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-14:
Seems to work fine for me:
[image]