Ultimaker/Cura #3277: Font Issue

RedRightArm, 2018-02-10:
**Application Version**

**Platform**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Appesteijn, 2018-02-10:
We made a test version, it includes some bug fixes and a possible fix for the garbled texts. We have tested this version, but not extensively.
The download location is from my personal could storage, since we don't have another way to distribute 'daily' builds (yet).
Could you try Cura 3.2.1: https://stack.mybacklog.info/s/MgLnKXLL0P9ro4M ?

uffe, 2018-02-11:
I got see problem as well, tried the (unofficial) 3.2.1 above and it unfortunately does not fix the issue. See https://www.dropbox.com/s/v9jf32s8amtbdwk/Screenshot%202018-02-11%2020.39.54.png?dl=0

RedRightArm, 2018-02-12:
Same here. I installed this newer version and still have the same font issue. It looks like there's a problem with the font files. Particularly the bold fonts.

Appesteijn, 2018-02-12:
Could you try: https://stack.mybacklog.info/s/F78VhDOiMOQ9Qay

uffe, 2018-02-12:
Tried it, but the non-bold fonts are still messed-up: https://www.dropbox.com/s/zphd5e2j6us6vd0/Screenshot%202018-02-12%2015.24.16.png?dl=0

Appesteijn, 2018-02-12:
Then I'm out of ideas...

RedRightArm, 2018-02-13:
This new download gave me the same issue... at first. @uffe, try this. Load the newest version that @Appesteijn sent. Open it. Switch from dark interface to light interface. Close and quit. Open. Switch from light interface to dark interface. Close and quit. Open. Switch from dark interface to light interface. Close and quit. When you open it this time, see if the fonts are fixed.

Also, don't pay too much attention to following my suggestion exactly. If you have it on light, switch to dark as the first step. If you have it on dark, switch it to light as the first step. The point is, switch it back and forth three times in a row and see if it resolves itself.

Doing this just worked for me. No idea why.

vegestoc, 2018-02-13:
That did not work for me. Cura 3.2 macOS 10.13.3

RedRightArm, 2018-02-13:
Yeah, I spoke too soon. The fonts came back all screwy again after a short while.
[image]

@Appesteijn I had no problem at all with fonts on v3.1. Is it still possible to download 3.1? There's something I'd like to try.

Appesteijn, 2018-02-13:
So switching the themes made it work for a short while? That is really weird.

Older versions can be found here: https://ultimaker.com/en/products/cura-software/list

uffe, 2018-02-13:
I tried switching back and forth between themes and quitting between each, no visible improvements

Appesteijn, 2018-02-13:
Could you try this version: https://stack.mybacklog.info/s/dr4Dwy2lr4tNE0q

uffe, 2018-02-13:
Sorry, still no difference. Is there any way I can wipe a font-cache or maybe help you with info?

RedRightArm, 2018-02-14:
I downloaded Cura 3.1 for Mac OS. No problem with the fonts at all. So I downloaded Cura 3.2. Font issue again, same as when I started this thread. So I dug into the app resources and replaced Cura 3.2 fonts with the fonts from Cura 3.1. When I opened Cura 3.2 the fonts were still all wacky. Whatever the issue is, it isn't coming from the fonts. Something in Cura 3.2 is telling it to display these fonts all weird.

Appesteijn, 2018-02-14:
Last time we had this, it was a problem that certain margins and/or widths were not rounded numbers and fonts were rendered on behind-the-comma pixels (that don't exist and won't get shown).

I went through all margins and widths to make sure they are rounded numbers, but that didn't solve the problem for you guys.
I also changed the 'custom mode' qml files to the state they were like in 3.1 but that also didn't solve it.

In this sprint we are changing the font (to NotoSans) and updating to Qt 5.10. If I have a build that has theses changes, I'll post it here so you can test it.

Appesteijn, 2018-02-14:
Last, last try: https://stack.mybacklog.info/s/iQBrFQrR3DxB7ol

RedRightArm, 2018-02-14:
The only font giving me an issue with this new version is the one that italicizes when you adjust a setting.
[image]

RedRightArm, 2018-02-14:
I spoke too soon. After closing and reopening a few times, the fonts went all wacky again.
[image]

nean-and-i, 2018-02-14:
3.2.1 font issue still there, even if it is highlighted to be fixed in the realease notes:
macOS 10.12.6
[image]

fieldOfView, 2018-02-15:
I am 90% certain PR https://github.com/Ultimaker/Cura/pull/3320 fixes the remaining problem.

Appesteijn, 2018-02-15:
Thanks to @fieldOfView it looks like we have a solution. Can you test this build please: https://stack.mybacklog.info/s/wOg49yivjI6AIHv ?

vegestoc, 2018-02-15:
[image]
Looks great here, thx everyone!

nean-and-i, 2018-02-15:
confirm its working now.
But is it normal that font rendering still takes ages on startup (meaning up to 20 sec until tree is shown) ?
[image]

vegestoc, 2018-02-15:
This is the fastest booting version of Cura I have used. 10 seconds 'loading machines' and an extra two seconds to load the tree menu. 

SPECS: 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3, NVIDIA GeForce GT 750M 2 GB, Intel Iris Pro 1536 MB. Might be worth mentioning that I have a lot of fonts installed.

RedRightArm, 2018-02-16:
Looks perfect at my end too. Thanks!

nean-and-i, 2018-02-16:
@vegestoc , very strange, my specs are: 3,3GHz I7, 24GB Ram, NVIDIA GeForce GTX 1070 8GB, macOS 10.12.6.
But, after startup time, only the first 3 lines of the tree are shown after 2 seconods or so the rest of the tree takes that long.

cura startup after first couple of seconds
[image]
... than it takes another 15-25 sec in that state until the rest of the tree is shown:
[image]

is there a way to debug this?

fieldOfView, 2018-02-16:
The best way to solve that delay is to NOT make all the settings visible. Learn to use the search on top instead of always having to scroll through the long, long list. The idea is - and always has been - to only make those settings visible you use a lot.

The delay happens because the GUI framework (Qt) is creating/drawing all the elements for all the settings. This predrawing can be turned off, but that causes delays while scrolling through the long list.

24c, 2018-02-18:
@Appesteijn "Can you test this build please:" Just a heads up, works fine after using your 3.2.2 macOS build above, and all text renders properly now. Thank you

nean-and-i, 2018-02-18:
@fieldOfView , indeed after getting rid of most of the fields its much faster now.
**BUT:**
I'd agree on using the concept of a search bar, but then one just can't see all the related options within a section of the keyword, and thats the reason to fallback on tree structure!

Maybe all options can be tagged internally and presented properly on a search, so the output is structured in a way one can even find all related options/parameters to a given search phrase (even if its name differs from search input)?

Otherwise it may sounds like a bit, the search bar exists only because getting rid of rendering issues,...

fieldOfView, 2018-02-18:
Search could be improved to take into account category names

ianpaschal, 2018-02-26:
Since this is fixed and will be included in the next release, I'm closing this issue.

closed 2018-02-26 by ianpaschal

fieldOfView, 2018-02-26:
Has a decision been made if the next release will be Cura 3.3 or Cura 3.2.2? If we have to wait until Cura 3.3, I may do a plugin to specifically fix this issue.

ianpaschal, 2018-02-26:
I am not sure, actually. Paging @Appesteijn.

Appesteijn, 2018-02-26:
3.3 Beta is planned in 3-4 weeks, that will be the next release.

mechaMecha, 2018-02-28:
I'd love to be able to download the Cura build that fixes the broken text problem, but it's no longer available through the link provided by @Appesteijn. Is it available anywhere else?

fieldOfView, 2018-02-28:
You can download and install my plugin here:
http://files.fieldofview.com/temp/PrepareStageCompactor.curaplugin

This plugin makes the sidebar more compact and fixes the fontissue as a sideeffect.

Appesteijn, 2018-02-28:
https://stack.mybacklog.info/s/PyEfKmkt63jkVZ3

mechaMecha, 2018-02-28:
Awesome! Thank you very much, @fieldOfView and @Appesteijn !

mechaMecha, 2018-02-28:
@fieldOfView , your plugin doesn't correct the problem for me, although it does work better when I use the dark interface. @Appesteijn , installed your Cura build and everything looks perfect. Thanks again!

SConaway, 2018-03-01:
@Appesteijn , your version works much better.

markwain, 2018-03-14:
@Appesteijn can you post the file again, it's no longer available ...

fcollingwood, 2018-03-30:
@Appesteijn can you please post the file again - the last link is no longer active

24c, 2018-03-30:
@fcollingwood Have you seen the 3.3 Beta, as this fixes the font issue on macOS too.

ianpaschal, 2018-03-31:
@fcollingwood We won't distribute that test build anymore as it's too difficult to have people running these one-off test builds rather than official builds and still diagnose problems. As @24c says though, 3.3 beta has been released which fixes this issue.

Andreaux, 2018-04-08:
@24c Thanks for the advice. I downloaded the beta and although the sidebar seems OK, the Plug-ins window now has garbled fonts. I didn't check further, there might be other issues. Annoying :(

24c, 2018-04-08:
@Andreaux Odd, I have that too, but never noticed that one at all, but it's almost the same as the Sidebar distortion previously. I wonder if their plug ins font is pointing to the old one, not the updated NoToSans version? There's probably an issue logged, but I've added to a thread over there. Thanks

fieldOfView, 2018-04-08:
The font is not the problem, the renderer is. It should be easily patchable. Thanks for reporting.

Andreaux, 2018-04-08:
@24c Thanks for your help. Hope it gets fixed.

kilomonk, 2018-04-08:
Cura 3.3.0 BETA works for me on 2012 MacBook Pro non-retina running the latest OSX Developer BETA  Version 10.13.5 Beta (17F35e)

Thank you.