Ultimaker/Cura #3280: Poor font rendering with High DPI scaling

codemonkeh, 2018-02-10:
**Application Version**
3.2.0

**Platform**
Windows 10 Pro x64

**Qt**
n/a

**PyQt**
n/a

**Display Driver**
nVidia GeForce 390.65

**Other particulars**
* Screen Resolution: 3840 x 2160
* Scaling: 150%

**Steps to Reproduce**
1. Adjust scaling to 150%
2. Open Cura
3. Open Preferences > Configure Cura

**Actual Results**
The text in the Print Setup and Preferences is sharp, however random characters in the preferences dialog are displayed in slightly different sizes / scaling. Further more they will 'dance', i.e. continually adjust, if you resize the dialog.

E.g. the *Interface* and *Viewport behavior* sub headings.

[image]

I do note that if you adjust to override the high DPI scaling to System or System (Enhanced), e.g.

[image]

then the text is correctly rendered, however it is slightly blurry (as i presume it is using bitmap scaling).

[image]

**Expected results**
The text should be sharp and scaled to a consistent size.

diegopradogesto, 2018-02-12:
We are aware of some issues with fonts, such as #3277. We are trying to fix them in the next release. Thanks for the feedback.

Appesteijn, 2018-02-12:
Could you try: https://stack.mybacklog.info/s/F78VhDOiMOQ9Qay

Appesteijn, 2018-02-13:
Could you try: https://stack.mybacklog.info/s/LDltcUP2BXur7AZ ?

codemonkeh, 2018-02-16:
Sorry @Appesteijn, I am getting a *Pagina niet gevonden* for that link

ianpaschal, 2018-02-16:
@codemonkeh I think that's because @Appesteijn replaced it with an updated version. Try:
https://stack.mybacklog.info/s/wOg49yivjI6AIHv

codemonkeh, 2018-02-22:
@ianpaschal / @Appesteijn This appears to be a macOS build, was there a windows build to test?

Apologies for the delay in my reply.

diegopradogesto, 2018-07-04:
@codemonkeh do you still see font rendering issues in our latest 3.4 version? Can you try it out?

closed 2018-07-31 by ianpaschal