Ultimaker/Cura #3281: Icon not applied to .STL files

lunaticcalm, 2018-02-10:
**Application Version**
Cura 3.2

**Platform**
Win10x64 RS3

**Display Driver**
Radeon 17.12.1

**Steps to Reproduce**
Install Cura 3.2

**Actual Results**
.STL files appear in File Explorer with the generic Windows 'unknown file type' icon

**Expected results**
.STL files appear with Cura's icon

**Additional Information**
[image]
Here's an example I found showing how it used to look in 3.1: 
https://content.invisioncic.com/ultimake/monthly_2017_10/2.5.png.7f983a40a0b6765a3254921ead3008d6.png
I've tried reinstalling Cura, changing the file association for .STL away then back to Cura, no luck.

Spankyty, 2018-03-03:
Same issue...

Ghostkeeper, 2018-03-28:
I've seen it work though... Let's investigate when this happens and when this doesn't happen.

SuddenlyPineapple, 2021-05-28:
Same problem in 4.9.1. Reinstall (without deleting settings) won't work. Change for other default app (like Paint 3D) apply other app icon and changing back to Cura switch to default windows icon :/ Seem to me like problem with paths to icon somwhere in code or when configuring file type extension <-> icon assotiation. I will try do it manualy and let know you ;)

SuddenlyPineapple, 2021-05-28:
I've found it. Seems like paths from previous intalation persist ;) 
[image]

This article helped find it ;) https://www.howtogeek.com/howto/12383/change-a-file-types-icon-in-windows-7/

GregValiant, 2024-12-15:
Has this been fixed in current versions (5.8 and up)?  Can this be closed?

closed 2024-12-30 by github-actions[bot]