Ultimaker/Cura #3282: master: Option missing to install plugins manually

thopiekar, 2018-02-10:
Just wanted to install a plugin manually via preferences, but found this:

[image]

Not only that the text is not readable, when the system theme is black, there is also no possibility to add plugins manually anymore. Seriously, how do you think should it be possible to distribute testing builds of plugins in the future like this?

[image]

Or has the decision really been made to prevent users to install plugins from 3rd-party locations?

fieldOfView, 2018-02-11:
See the discussion here; https://github.com/Ultimaker/Cura/pull/3203#issuecomment-363185975

@ianpaschal: The text color in the first screenshot is because of the use of Text instead of Label. Label is defined as Text with a preset color (and some other properties) that defaults to the system text:
https://github.com/pyqt/python-qt5/blob/master/PyQt5/qml/QtQuick/Controls/Label.qml#L79

thopiekar, 2018-02-11:
@fieldOfView Thank you for pointing out this discussion :+1:

Vandrasc, 2018-05-07:
Fix included in the next release.

closed 2018-05-07 by Vandrasc