Ultimaker/Cura #3283: Cura Materials Disappear when changing filament size

madebyjeffrey, 2018-02-10:
**Application Version**
3.2

**Platform**
Windows 10

**Qt**
N/A

**PyQt**
N/A

**Display Driver**
Intel HD 5000  20.19.15.4835

**Steps to Reproduce**
Create a new material and fill in all the settings prior to filament size. 
Enter the filament size.

**Actual Results**
Material seemingly disappears.

**Expected results**
Material doesn't disappear.

**Additional Information**
What is actually happening is that the material is being removed from the default machine filament list (2.85) to the list for 1.75 without any notification. I was setting up a new machine definition and didn't see the part about the extruder.

When I changed the the extruder setting to 1.75, it showed up along with the other three I created because I didn't know what was happening until then.

This is an issue for anyone that is new to Cura and doesn't notice it. When it moves it, it should notify the user in some way.

fieldOfView, 2018-02-11:
I distinctly remember there was a notification message before.

Hmm, briefly looking at the code show that that message may only be showing in the main window (which could be hidden from view by the preference dialog). Devs, see MaterialManager.showMaterialWarningMessage()

diegopradogesto, 2018-02-13:
I'm taking a look at it. Thanks @fieldOfView for the advice.
Take into account that the message is only shown when the diameter changes and the printer defintion has defined the `has_machine_materials` to true.

diegopradogesto, 2018-02-13:
Devs, I created CURA-4947 but the issue is not related with the material disappearing but the lack of a message indicating that fact.
@madebyjeffrey the behaviour is the expected, but I agree that a message must be shown.

diegopradogesto, 2018-02-13:
Fixed in [9053282](https://github.com/Ultimaker/Cura/commit/9053282e8ae0d667c18f0bddd8024181fe50fa9f). Waiting for colleagues to review and test.

closed 2018-02-13 by diegopradogesto