Ultimaker/Cura #3284: Cura 3.2: wrong filament length displayed, correct length in gcode comments

cmock, 2018-02-10:
**Application Version**
Cura 3.2 (downloaded from website)

**Platform**
Debian GNU/Linux 9

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
- created a new printer profile, with filament diameter set to 1.75mm
- sliced a big model
- noticed the displayed filament length of 119m doesn't match with the weight of about 1kg
[image]
- in the generated gcode file, the length seems correct:
``;Filament used: 315.881m``
- switching to another printer profile, also with 1.75mm filament, the displayed length is about 315m.

**Actual Results**
Filament length displayed is wrong.

**Expected results**
The correct length should be displayed

**Additional Information**
The ratio of wrong to correct length is the same as the area ratio of 1.75mm vs 2.85mm filament.

Restarting cura doesn't help.

The printer profile showing the problem was freshly created with Cura 3.2, while the profile which shows the correct length was imported via the upgrade.

The selected material also has a diameter of 1.75mm configured.

fieldOfView, 2018-02-11:
It would be my guess this is because material diameter has become a per-extruder setting, but this one calculation still use the value from the “global” stack

Ghostkeeper, 2018-03-05:
Fixed for 3.3. Thanks for the report!

commit 32fedf95 (2018-03-05): Obtain filament diameter from stack for length calculation

Otherwise we get it from the global stack which is always 2.85mm.

Fixes #3284.
closed 2018-03-05 by Ghostkeeper