Ultimaker/Cura #3288: Cura 3.2.0 - Layer View feedrate display does not honor post processing speed change

nicolinux, 2018-02-11:
**Application Version**
3.2.0

**Platform**
macOS 10.12.6

**Qt**
bundled

**PyQt**
bundled

**Display Driver**
macOS default

**Steps to Reproduce**
Add the Tweak-at-Z post-processing plugin and change the speed for a number of layers.

**Actual Results**
The layer view feedrate display still displays the default feedrate color for the overall printing speed.

**Expected results**
Layer color should change according to feedrate. It is not clear here if the post-processing is not enabled or working correctly.

**Additional Information**
Screenshot:
[image]

fieldOfView, 2018-02-11:
None of the postprocessing scripts act on layerview data. This is an architectural choice. Postprocessing scripts are not executed until the gcode file is saved. Currently the only way to see the effect of postprocessing is by exporting the gcode and loading it back in.

jackha, 2018-02-12:
What @fieldOfView says... discussed it and we're not going to fix this.

fieldOfView, 2018-02-12:
> discussed it and we're not going to fix this.

I might give it a go.

fieldOfView, 2018-02-12:
> discussed it and we're not going to fix this.

I might give it a go, by allowing postprocessing scripts to also postprocess the layer data.

diegopradogesto, 2018-02-12:
The only way to see this now is by exporting the data to gcode and then import the gcode again. That should work.

GregValiant, 2026-09-08:
I'll go ahead and close this.  There is a workaround that involves opening the saved Gcode file in Cura.

closed 2026-09-08 by GregValiant