Ultimaker/Cura #3289: Cura doesn't really uninstall older version when installing newer and messes file type up

MathijsG, 2018-02-11:
**Application Version**
3.2 but others did this too
**Platform**
Windows 7 Home Premium

**Steps to Reproduce**
Install a newer version of Cura and let this installer remove the older one. It says it finishes, but in the end the older Cura still stays installed here. The actual problem is that somehow after manually uninstalling (by clicking the uninstaller file in the Cura folder of the older Cura installation) the file types for STL file becomes messed up, and when I manually browse to the newer Cura it still doesn't change that to be the associated program again.

**Actual Results**
Older Cura is still installed when newer one is installed, while the installer says it removes the older one.

**Expected results**
Older Cura being removed, file assocation being updated with the latest Cura.

**Additional Information**

alekseisasin, 2018-02-12:
I tried to uninstall (Cura 2,7, 3.0, 3.1) during installing Cura 3.2 and all old folders of Cura where removed. For now I tested on windows 10,
If for some reason the Cura was not uninstalled and you uninstall it after installation then I assume it also removes the program link which should be associated for a specific (in our case .stl) file extension.

MathijsG, 2018-02-12:
The biggest problem is in fact that you can't easily add the file association back again if this is somehow messed up after install. Maybe we can provide something in settings that changes the file association again?

Appesteijn, 2018-02-12:
Can you try to reinstall 3.2 ?

MathijsG, 2018-02-12:
I did, and then it was fixed. But to me it's not an acceptable solution to manually reinstall Cura everytime you somehow changed the file associations :-).

Appesteijn, 2018-02-12:
No of course not :) But then the real error is that the uninstall went wrong, right? Could you provide use with more info on why the uninstall could have gone wrong?

MathijsG, 2018-02-12:
Not yet, I will try further. This is not isolated, because it was the same problem every time earlier.

But is it correct that if you didn't select Cura to be the default STL program at the end of the installation, it SHOULD set it to Cura if you would manually select the path to Cura.exe in the Windows settings for default programs? Or is there a more complex path needed for this, because of the python structure?

Because maybe that's the real issue I encounter here: If I manually select Cura to be the default STL program associated to *.stl then it somehow plainly ignores this selection and sets it back to the previous selected one (S3D in my case) after clickingApply.

Ghostkeeper, 2018-03-07:
Yeah, setting the file association in Windows yourself to Cura.exe should work.

If that didn't register in Windows somehow that would be a bug in Windows.
I mean, you could rename a Word document with your thesis to .exe and associate STL files with that; Windows should register the file type to be opened with that .exe even if it clearly won't work.

ianpaschal, 2018-03-27:
Relates to #3281.

closed 2018-03-27 by ianpaschal