Ultimaker/Cura #3290: Cura crashes while adding monoprice printer

cf5000cf, 2018-02-12:
Cura 3.2.0 crashes while adding any Monoprice printer on Windows 10

jackha, 2018-02-12:
The issue should be fixed in the 3.2 release version.

Is it the 3.2.0 or the beta? Did you use the beta? In that case you should delete the cache directory: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

Duplicate of #3246

closed 2018-02-12 by jackha

syadnom, 2018-03-08:
3.2.1 still does it.

jackha, 2018-03-08:
I started clean (in linux, but that shouldn't matter), added both the Monoprice V1 and V2 without problems.

Did you clean your user folder after crashing? Does it still crash? If so, please share the logs with us so we can have a look at it.

syadnom, 2018-03-08:
Started 100% fresh.  I can only get the mini delta's profile to run in 3.0.4, which is acknowledged on the monoprice download site for the profile.