Ultimaker/Cura #3292: Acceleration, bad code generated

povlhp, 2018-02-12:
Running 3.2, and I have issues with the genrated gcode when using acceleration from within Cura.

I get code like this:
<code>M204 S2400
M205 X15 Y15
G1 F2700 E60.37217
G1 F300 Z1.3
G0 F3600 X84.044 Y111.972 Z1.3
M204 S2000
M205 X12 Y12
G1 F300 Z0.3
G1 F2700 E65.3888
G1 X87.027 Y111.972 E65.53018
M204 S2400
M205 X15 Y15
G0 F3600 X87.978 Y111.424
</code>

Since I am using Marlin, it is possible to set the values once per file, using specific values. See http://marlinfw.org/docs/gcode/M204.html

M204 P2000 T2400 

Would set travel to 2400 and print to 2000mm/s^2

As it is, curves are printed very slowly (I am at 115kbps, as I have enabled Linear Advance in FW). With the current implementation, I will not use it. Then it is way better to put the values in header gcode or early on like setting initial layer temps.

smartavionics, 2018-02-12:
It would be trivial to make the change you suggest but according to http://reprap.org/wiki/G-code#M204:_Set_default_acceleration, Marlin hasn't always supported the P and T values so I don't know if this can be done without causing a problem with backwards compatibility.

povlhp, 2018-02-16:
I was looking at the Marlin source. And even the 1.0.x branch has support for P and T values:
https://github.com/MarlinFirmware/Marlin/blob/1.0.x/Marlin/Marlin_main.cpp (latest commit august 2015)

See line 2903 and forward.

The oldest 1.0.x github version from November 6th 2012 has it as well.
https://github.com/MarlinFirmware/Marlin/blob/06b58a9c4f6c9b01a36a17e264652891850a85eb/Marlin/Marlin_main.cpp

So I think it is safe to do the change of gcode if we are using Marlin/RepRap as gcode dialect. So it has always been supported by Marlin.

povlhp, 2018-02-16:
smartavionics I mistakenly looked for M205. M204 with P/T was not implemented before 1.1, so doing the change will break backward compatibility.

smartavionics, 2018-02-16:
OK, understood. Well, if this can't be done, could you please close the issue? Thanks.

povlhp, 2018-02-16:
Closed. Would break backward compatibility.
For Marlin 1.1 and never, set jerk and acceleration in start gcode.

closed 2018-02-16 by povlhp