Ultimaker/Cura #3293: Variable layer height not respecting top thickness

badvision, 2018-02-12:
Loving the new variable layer height feature but I'm printing several pieces at once and some parts are getting pillowing because I think the layers at that part were thin and it picked the same number of layer it would have otherwise picked, basically there wasn't enough material extruded in those layers to cover the infill correctly but I get good results in other parts of the print.  Should this feature instead just go off of top /bottom thickness and ignore # of layers?

jackha, 2018-02-13:
Can you show a case where it didn't work out as you hoped? I think especially for round like parts the top and bottom are important.

ianpaschal, 2018-02-26:
Reporter has not provided more information for more than 2 weeks, so closing.

Ghostkeeper, 2018-03-16:
This still reproduces. It's actually quite clear. Here's what happens:
* The user specifies that he wants a top thickness of 1mm.
* Cura (front-end) calculates based on the layer height that this means that the user wants 10 top layers.
* Because of Adaptive Layer Height, CuraEngine changes the layer height in this zone to 0.04mm.
* CuraEngine draws 10 top layers at 0.04mm.
* The top thickness becomes 0.4mm instead of 1mm.

This is in my opinion one of the most serious issues with adaptive layer height because the feature quite often makes the top layers much smaller than the normal layer height.

Ghostkeeper, 2018-03-16:
I'm thinking that maybe we should disable the "Top Layers" setting that allows the user to specify the number of layers as a count, when adaptive layers is enabled, and then use the specified height instead, so that you get more top layers in areas where the layer height is smaller.

treymd, 2018-04-30:
I was about to report this bug as well, but it is already here.  I have my top thickness set to 1.6mm and adaptive layers turned on.  Cura ends up printing 8 layers at 0.1 layer height instead of the 0.2 default the print is set at, cutting the thickness of the top layer in half.  Just confirming that this is indeed happening.

mechaMecha, 2018-05-20:
Experiencing the exact same problem. Definitely a bug. For now I need to double or triple the top thickness setting to get acceptable top layers.

mechaMecha, 2018-05-20:
This isn't even a difficult bug to reproduce. All you need to do is take a cube, fillet the top surface, then slice it in Cura with adaptive layers turned on. Slice it using 1.5 mm as the base layer height, 0.1 as the maximum variation, and top thickness set to 1 mm. There will be 7 top layers generated, and each layer will be approximately .05 mm thick. That results in a top surface that is approximately .35 mm thick, not anywhere close to the specified 1 mm.

SmileyNeil, 2021-05-16:
Same issue applies to bottom layers.

Ghostkeeper, 2021-05-31:
As for an update, this issue is still in our backlog. Developers, refer to issue CURA-5104.

It is considered a requirement for moving Adaptive Layers out of experimental.

smartavionics, 2021-05-31:
See https://github.com/smartavionics/Cura/releases for a Cura variant that implements what is being asked for here (i.e. the thicknesses of the top and bottom skins are maintained when variable layer heights are enabled).

timperlabs, 2022-04-09:
This is still happening in Cura 4.13.  

It might be useful to change the popover tip for this feature to include known limitations so that new users don't have to spend much time re-discovering why their rounded top surfaces have holes or irregularities.