Ultimaker/Cura #3294: Printing via USB bugs

Machine4000, 2018-02-12:
**Application Version**

3.2

**Platform**

Windows

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

1) --> When printing via USB and you cancel the print, you cannot re-print unless you restart cura application.  I am using printrbot simple metal printer here.

2) --> The temperature display does not increase while the printer is warming up when you prive via USB, it is not helpful to see if the printer is doing anything or not.

3) --> One another note, it would be very helpful to have a restart print button and manual controls for the extruder to fix blockages.

**Actual Results**

**Expected results**

**Additional Information**

sfornengo, 2018-02-14:
I can confirm all this problems too with my printrBot simple metal, on ubuntu 16.04, cura appImage 3.1 and 3.2

ChrisTerBeke, 2018-02-15:
USB printing is not really supported. It works, but we advise against using it. It's a flawed way of controlling the printer as anything can happen to the USB connection and disrupt your print. Also firmware support in most printers leaves a lot to be desired.

Ghostkeeper, 2018-03-28:
USB printing is supported. For some of Ultimaker's printers that is the only way in which you can print at all. It's not supported for the Ultimaker 2, maybe you got confused there.

Ghostkeeper, 2018-03-28:
1. That sounds like a bug indeed. I recall that it's been reported before but I can't find it now. I just tried on an Ultimaker Original, but I cannot reproduce the bug. Maybe your printer doesn't respond properly somewhere?
2. The temperature display should still work. It does seem to work fine in my test just now. However the target temperature won't display correctly while it's heating. The reason for this is because we can only sample that temperature by sending a g-code command via USB. While the printer is heating it's executing a single g-code command to do so, and the command queue is blocked until heating is complete. It won't execute the g-code commands we send to it until heating is complete so we don't get a response with the new target temperatures.
3. Also could not reproduce this with an Ultimaker Original. I believe that Cura should not display such weird behaviour regardless of how the printer responds, but for us it's hard to fix this because we don't know how to trigger that bug without having a Printrbot Simple Metal.
4. Manual controls are in there. A restart print button could be implemented though. Only problem is that it would take a whole lot of extra memory.
5. Is the progress bar not sufficient?

ChrisTerBeke, 2018-03-28:
Getting mixed signals here. Some people say it's supported, others say it's not...

USB printing is not (?) support on Ultimaker printers (or at least not preferred), so I guess it's somewhat supported in Cura but more a community effort?

fieldOfView, 2018-03-28:
There are still a good number of Ultimaker Originals around from when the UltiController was an addon.

sfornengo, 2018-03-28:
pity that the communication was not clear (and still is not), announcing that what suddenly did not work would not be fixed, you forced many users like me to leave cura and go to an other solution like simplify3D...

lee-b, 2020-01-05:
> USB printing is not really supported. It works, but we advise against using it. It's a flawed way of controlling the printer as anything can happen to the USB connection and disrupt your print. Also firmware support in most printers leaves a lot to be desired.

Frankly, that's a ridiculous position to take.  Makes me want to fork the project (if only I had time).

There's absolutely no reason to blame USB, when slic4r, pronterface, simplify3d, etc. all work fine.  This is either lack of interest in non-Ultimaker hardware (which you should admit to, if it's the case, so we can all stop contributing to Ultimaker and focus on genuine community-oriented software) OR it's denial (which you should admit to for everyone's sake, including your own).

Ghostkeeper, 2020-01-13:
Some clarifications for you, then:
* We are not blaming the USB printing protocol. It really has nothing to do with this.
* Ultimaker's stance on USB printing is that it's supported to print via USB cable to an Ultimaker Original. It's not supported to print via USB cable to any other printer; we provide no guarantees.
* Printing via USB cable to other printers may still work even though we don't actively support it. I know for a fact that it still works with an Ultimaker 2 since that's how I usually test it unless I really suspect that the bug is on the firmware or protocol side.
* Ultimaker has no interest in spending money to fix issues with USB cable printing to printers that are not supported. For its own printer models it's not the advised way to print. For other printers there is no realistic way for us to test the fixes and no incentive either.
* The only printer supported by Ultimaker, the Ultimaker Original, is no longer being sold. So Ultimaker is maintaining the current workflow but no longer developing new features for it.
* Ultimaker will merge external fixes and features to USB printing if someone else develops them. So feel free to make a fix and propose it to us. We'll check if the fix doesn't break anything in the supported USB workflow (Ultimaker Original) and if it doesn't it should be fine.
* We don't admit or deny that these bugs exist. We simply can't test it.

Feel free to release your own version of Cura. However if you do, you need to adhere to Cura's license which means that we're allowed to pull your fixes into our own version of Cura if we want. If another company wants to develop Cura to work better with their own printers then we'll be glad to merge their fixes in our version of Cura, as long as they don't block our own workflows and they are maintainable in the long term.