Ultimaker/Cura #3295: Tiny [FR] Show version number in the window title ,e.g.,  "Ultimaker Cura 3.2"

filipgoc, 2018-02-13:
**Current State**
[image]

**Desired**

 "Ultimaker Cura 3.2"

**Rationale**

For those of us who run multiple versions of Cura, which should be anyone who is testing any beta for example, it is sometimes confusing to know which version we have currently opened. Sometimes I have multiple opened when I'm comparing behavior, too... 

My current workaround is to edit the graphics files to get instant version recognition, but that feels hackish of course. 

[image]
[image]

ianpaschal, 2018-02-13:
I don't know of a single Mac application with this behavior, and I don't think we would add it quite like you describe. A compromise, though, might be to include version numbers on beta versions instead of stable releases.

jackha, 2018-02-13:
I like the graphics ^^

ianpaschal, 2018-02-14:
@LukeChen-Ultimaker

filipgoc, 2018-02-14:
Yeah, maybe it's not worth it. For the beta versions, it could be worth it. No biggie.

fieldOfView, 2018-02-17:
I have created a plugin for you:
https://github.com/fieldOfView/VersionInTitlebarPlugin

Easiest way to install this is to download the curaplugin here:
http://files.fieldofview.com/temp/VersionInTitlebarPlugin.curaplugin
Drop it in Cura and restart Cura.

filipgoc, 2018-02-17:
You're such a beast @fieldOfView !