Ultimaker/Cura #3296: Multiply selected models/reset all model positions incorrect

sgetteman, 2018-02-13:
**Application Version**
3.2.0

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

1. Created a  slim 2in x 2in x 1/8in thin model positioned on edge
2. Right mouse click and select multiply selected models
3. I entered 15 (total 16 desired) and it correctly replicates but not in alignment
4. Right mouse click and select reset all model positions to align in a row
5. The number of selected models that begin to align changes from the number multiplied above
6. If you select reset all model positions again the models pictured changes again to a different count
7. Each time you right click the reset all model positions the number of displayed models changes
8. Also, I am not sure if related but I noticed in layer view the models are not built sequentially. I would expect model one layer one then model two layer one, etc.  What I see is model one layer one then maybe model four layer one, model five layer one, then back to model two layer one, model six layer one, etc.  Is this normal?  I can't seem to re-adjust the sequence to cause sequential printing.

I have tried multiple times and each occurrence produces a different result

**Expected results**
I would expect when printing multiple models in alignment that the prints would be sequential and not jump back and forth between models

**Additional Information**

jackha, 2018-02-13:
I think you're printing in One At A Time? The expected results suggest that.

Can you verify the following? It looks like the models are always printed in the same order. So, if you switch two models then it switches the order as well?

sgetteman, 2018-02-13:
Hi Jack, 

I'm printing in recommended mode, not custom. I checked special modes and it shows all at once is set. 
I may have found a solution to the display. If I setup 16 total (or any multiple models) and align them, there are only 12 displayed (in this example) but bunched together. If you simply move them apart individually, all 16 will finally appear. All the models are physically there, some just hidden. Also, after watching the build sequence in layer view, I move those that are out of order to be in order and the build still jumps around from model to model. Is there something magical I need to understand about alignment? 

Thanks for your help. 

Steve

jackha, 2018-02-14:
So if I understand correctly, you mean that it "jumps around" on a per layer basis? There should be nothing magical about alignment; the travels from model to model in each layer is calculated automatically. 

The multiple objects on top of each other suggest that the "Ensure models are kept apart" is turned off? But I'm a bit puzzled, because multiplying objects should find new locations for each object. 

Is it possible to share a project or screen captures to show what you mean?

sgetteman, 2018-02-14:
Attached is another example. 

1. Loaded a board tile from a chess set stl file. 
2. Rotated to vertical. 
3. Right mouse clicked and multiplied by 15 to get 16 total squares 
4. Cura shows 16 (see snip 1) 
5. Right mouse click on Reset all model positions 
6. Cura shows 10 in gray shadows (see snip 2). 
7. At this point Cura looks like it is trying to expand to 16 as I can see the gray shadows moving among themselves? 
8. Shows ready to slice (but it is not)...notice layer view does not show the actual models 
9. Separate each model by moving each away from the others (see snip 3) and you can see all 16 models. 
10. Now slicing will take effect. 
11. Note in layer view the sequencing of the models (left to right 1 to 16). 
12. The left most being model 1 layer 1 is printed first, then model 5 layer 1, then model 9 layer 1, etc. 
13. You can see it is not sequential from 1, 2, 3, 4, to 16, correct? In this example it is 1, 5, 9, 15, 14, 13, 12, 11, 10, 8, 7, 6, 4, 3, 2. 
14. Although the prints jump from model to model the print layers appear correct. If I were to right mouse click on Reset all model positions again, the sequencing will be different from what I just described. I tried varying counts of models and found the same similar reults. 
15. I verified, and Ensure models are kept apart is checked ON. 
16. Hopefully you can reproduce my findings. 

Thanks again for working with me. 

Steve

jackha, 2018-02-15:
Can you attach in github? Replying in an e-mail with attachment doesn't work.

sgetteman, 2018-02-15:
Screen snips 1-4 follow for the above steps.

[image]
[image]
[image]
[image]

sgetteman, 2018-02-15:
Latest screen shots included per your request.

jackha, 2018-02-19:
Ahhh now I understand what you mean. "Reset All Model Positions" is actually flawed at the moment, it puts all models to the (0, 0) location, which makes no sense. Then the "automatically push away" pushes them away and you see the dancing. Saving the initial location after adding the models and reset to that location later makes more sense. I'll talk to the team to see what to do with it.

Using "Arrange all models" is better and results in a similar setup as just after the multiplying.

Ghostkeeper, 2018-03-09:
The screen shots help.

We'd expect the models to be positioned all next to each other if moving away from each other is enabled in the preferences. That seems to be buggy.

We'd also expect the nozzle to continue with the closest other polygon after finishing one of those rectangles, so the order would be from one side to the other each layer. That is also incorrect.

Ghostkeeper, 2018-03-09:
I'm unable to reproduce the first bug (arranging doesn't move everything away from each other). I created 16 cuboids of 40x4x1mm. They were arranged in a similar fashion as in Snip 1. Then I selected Reset All Model Positions and they were put on top of each other, and subsequently properly arranged in one long line, all 16 apart.

I did reproduce the ordering bug though. It printed them in a seemingly random order.

GregValiant, 2024-11-14:
There have been many changes to Cura since this was opened.  Can this be closed?

closed 2024-12-10 by github-actions[bot]