Ultimaker/Cura #3300: Cura 3.2.0 - New build plate feature overwrites same file when saving gcode

nicolinux, 2018-02-13:
**Application Version**
3.2.0

**Platform**
macOS 10.12.6

**Qt**
bundled

**PyQt**
bundled

**Display Driver**
macOS default

**Steps to Reproduce**
Using the new feature in 3.2.0 one can slice several objects and assign them to a new build plate. This is great but when gcode is saved to file (slicing for an UM2+), the same file name is used for consecutive saves.

**Actual Results**
Two objects, object 1 assigned to build plate 1, object 2 assigned to build plate 2. First object file name is "test.stl", second object file name is "hello.stl".
Select build plate 1 and prepare slicing. Then save to removable drive. The file "test.gcode" is created.
Select build plate 2 and prepare slicing. Then save to removable drive. Same file "test.gcode" is created and overwrites the first one.

**Expected results**
Each time "save to removable drive" is used, the name of the .gcode file should reflect the name of the .stl file assigned to the actual build plate.

jackha, 2018-02-13:
Hi @nicolinux , thanks for sharing! It is true that saving to file is not fully supported yet. Ideally x files are saved for x build plates, possibly with _0, _1, etc. 

Also when sending a print over the network, that's not fully supported yet:
- On legacy firmware (no Cura Connect), it will only send the current build plate
- On new firmware it sends all the build plates, but you cannot choose.

nicolinux, 2018-02-13:
I used it on the UM3 with the new firmware and found the workflow intuitive. It just printed the build plates one after the other - like I would expect it.
The new build plate feature just fails for "non networked" UM printers. I think an easy fix would be to just have different .gcode names.

jackha, 2018-02-13:
We've actually refactored the UM3networking stuff. In master it does not send every build plate yet, but it would be nice ^^

jackha, 2018-07-31:
Build plate names fixed in: e557751

closed 2018-07-31 by jackha