Ultimaker/Cura #3301: GCodeViewer doesn't show extrusion lines using relative extrusion slicing option

MB3Y, 2018-02-13:
**Application Version**
R3.2.0

**Platform**
Win10

**Steps to Reproduce**
- Slice a model using "Relative Extrusion"
- Safe the generated gcode
- File, Open your freshly generated gcode file

**Actual Results**
The model is badly represented - most of the extrusion lines are missing

**Expected results**
Model should appear correctly

**Additional Information**
This might be related to https://github.com/Ultimaker/Cura/issues/3190
Same gcode file seems correctly represented in http://gcode.ws/
[TriangleError.zip](https://github.com/Ultimaker/Cura/files/1721419/TriangleError.zip)
[image]

diegopradogesto, 2018-02-14:
We are aware of this. We introduce different flavors in the GCodeReader not so far ago and those are not fully implemented yet. We already know that we should implement it.
Thanks @MB3Y for the feedback.

Ghostkeeper, 2018-05-01:
This has been fixed in current Master according to our testers.

closed 2018-05-01 by Ghostkeeper