Ultimaker/Cura #3302: variable layer height + spiralize causing blobs.

UltiArjan, 2018-02-13:
Cura 3.2
Win10

**Actual Results**
When using variable layer height in combination with vase mode (spiralize) the head stalls for a short time at some layers, looking at the pattern there's probably a relation with layer height changes. It only occurs where the object is getting wider (see picture)

**Additional Information**
did a control print of the same object , with spiralise, but without various layer height, and the blobs did not occur.

[image]

ChrisTerBeke, 2018-02-13:
Does this only happen when using variable layer height? We've seen this before (and still) and there's some buffering issues in the UM3 sometimes when there are a lot of short movements.

UltiArjan, 2018-02-13:
It did not happen on the control print and the pattern makes me think it
does have to do with variable layer height.

<https://github.com/Ultimaker/Cura/issues/3302#issuecomment-365426043>, or mute
the thread

smartavionics, 2018-02-13:
I'll take a look at the spiralize code tomorrow to see if there is anything there that could be broken by variable layer heights.Will report back...

smartavionics, 2018-02-14:
I can't see any obvious problem in the slicer code so can you please make the gcode that has the blobs available for me to take a look at? Thanks.

UltiArjan, 2018-02-14:
sure, here's the project file and gcode (UM3) had to do it again as I did not save it, printing now, blobs at exactly the same spots again.

[VariVase.zip](https://github.com/Ultimaker/Cura/files/1724788/VariVase.zip)

smartavionics, 2018-02-15:
I think the problem here is one of extruder resolution. If you look at the gcode for the layers which have blobs you can see that some extremely small extrusion amounts are being output. Here's an example of the gcode, look at the E values:

````
G1 X116.5 Y94.419 Z1.739 E43.7537
G1 X117.631 Y94.417 Z1.741 E43.76176
G1 X118.285 Y94.488 Z1.742 E43.76646
G1 X118.877 Y94.583 Z1.743 E43.77073
G1 X119.478 Y94.706 Z1.744 E43.77511
G1 X120.076 Y94.859 E43.77951
G1 X120.702 Y95.051 Z1.746 E43.78418
G1 X121.329 Y95.279 Z1.747 E43.78894
G1 X121.931 Y95.534 Z1.748 E43.7936
G1 X122.53 Y95.822 Z1.749 E43.79834
G1 X123.109 Y96.138 Z1.75 E43.80305
G1 X123.68 Y96.489 Z1.751 E43.80783
G1 X124.244 Y96.875 Z1.752 E43.8127
G1 X124.798 Y97.298 Z1.753 E43.81767
G1 X125.327 Y97.752 Z1.754 E43.82265
G1 X125.829 Y98.227 Z1.755 E43.82757
G1 X126.284 Y98.706 Z1.756 E43.83229
G1 X126.715 Y99.209 Z1.757 E43.83701
G1 X127.115 Y99.727 Z1.758 E43.84168
G1 X127.538 Y100.335 Z1.759 E43.84696
G1 X127.87 Y100.872 Z1.76 E43.85146

````

Some of those extrusion deltas are down to 4um! I very much doubt the UM3 extruder can resolve that. So, what's happening (I think) is that where the individual extrusion lengths are very short, the printer will not move the extruder on until the cumulative distance has passed some threshold which is presumably related to the extruder's steps/mm value. When the extruder moves on, you get the blob.

I note that you are using quite a narrow line width, if you increase the line width, the amount extruded per mm will be higher and so it should reduce the blobbing.

If Cura was smarter it would take into account the max/min extruder rates that the machine was capable of and would not try to extrude outside of those limits. This is going to cause problems for people using the variable layer heights. Hope this makes sense!