Ultimaker/Cura #3303: master: Crash when opening plugin browser

thopiekar, 2018-02-13:
Using my master PPA together with PyQt5 5.9.2 and Qt5 5.9.3.
When opening the plugin browser, Cura crashes.
@ianpaschal: I guess you are interested in the log:

```
2018-02-13 23:06:46,063 - DEBUG - [(139938426689344)-MainThread] UM.Qt.QtRenderer._initialize [184]: Support for Vertex Array Objects: True
2018-02-13 23:06:46,069 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
2018-02-13 23:06:46,069 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.1.0 NVIDIA 384.111
2018-02-13 23:06:46,070 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   NVIDIA Corporation
2018-02-13 23:06:46,070 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: GeForce GT 720/PCIe/SSE2
2018-02-13 23:06:46,070 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/default.shader]...
2018-02-13 23:06:46,072 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/selection.shader]...
2018-02-13 23:06:46,073 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/default.shader]...
2018-02-13 23:06:46,076 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/composite.shader]...
2018-02-13 23:06:46,078 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/overhang.shader]...
2018-02-13 23:06:46,080 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/striped.shader]...
2018-02-13 23:06:46,083 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/transparent_object.shader]...
2018-02-13 23:06:46,084 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/striped.shader]...
2018-02-13 23:06:46,098 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/default.shader]...
2018-02-13 23:06:46,100 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/grid.shader]...
2018-02-13 23:06:46,102 - DEBUG - [(139938426689344)-MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/platform.shader]...
2018-02-13 23:06:47,490 - DEBUG - [(139938426689344)-MainThread] cura.Settings.MachineManager._checkStacksHaveErrors [439]: Checking 3 stacks for errors took 1.22 s
2018-02-13 23:06:48,083 - DEBUG - [(139938426689344)-MainThread] CuraEngineBackend.CuraEngineBackend.slice [195]: starting to slice!
2018-02-13 23:06:48,084 - DEBUG - [(139938426689344)-MainThread] CuraEngineBackend.CuraEngineBackend.slice [212]: Going to slice build plate [0]!
2018-02-13 23:06:48,084 - DEBUG - [(139938426689344)-MainThread] CuraEngineBackend.CuraEngineBackend.slice [216]: Build plate 0 has no objects to be sliced, skipping
2018-02-13 23:06:56,421 - DEBUG - [(139938426689344)-MainThread] AutoSave.AutoSave._onTimeout [71]: Autosaving preferences, instances and profiles
2018-02-13 23:06:57,602 - INFO - [(139938426689344)-MainThread] PluginBrowser.PluginBrowser.requestPluginList [132]: Requesting plugin list
2018-02-13 23:06:57,610 - DEBUG - [(139938426689344)-MainThread] PluginBrowser.PluginBrowser._createDialog [139]: Creating dialog [PluginBrowser.qml]
2018-02-13 23:06:57,730 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-13 23:06:57,731 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-13 23:06:57,731 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:   File "/usr/lib/cura/plugins/PluginBrowser/PluginBrowser.py", line 290, in pluginsModel
2018-02-13 23:06:57,731 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:     self._plugins_model = PluginsModel(self._view)
2018-02-13 23:06:57,732 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:   File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/PluginsModel.py", line 16, in __init__
2018-02-13 23:06:57,732 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:     super().__init__(parent)
2018-02-13 23:06:57,732 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:   File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 14, in __init__
2018-02-13 23:06:57,732 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]:     super().__init__(parent)
2018-02-13 23:06:57,733 - CRITICAL - [(139938426689344)-MainThread] cura.CrashHandler.__init__ [67]: TypeError: QAbstractListModel(parent: QObject = None): argument 1 has unexpected type 'str'
```

ChrisTerBeke, 2018-02-13:
Doesn't seem related to PyQt though

ianpaschal, 2018-02-13:
@thopiekar This is related to some changes I made today. Have you pulled the most recent Uranium as well?

ianpaschal, 2018-02-14:
Yes. This was a combined change in Cura and Uranium that required adding another argument to a function in Uranium. In Cura `PluginModel("installed")` was changed to `PluginModel(None, "installed")`.

ianpaschal, 2018-02-14:
No my commit is in both places. Thankfully whatever causes this is not that silly haha.

I will check tomorrow but I was not having such issues with the PPA. I believe I know what's causing it.

Also, running Qt 5.9 is not advised. I did for a while when I started at Cura because that was the latest version and it caused tons of problems. For this reason we avoided upgrading from 5.8.2 until today when we switched to 5.10.

thopiekar, 2018-02-14:
@ianpaschal haha, another secret PPA user? 😋 

I know, but that's the idea of the PPA.. Using as many dependencies as provided by Ubuntu itself.
Yeah, I saw the upgrade. But I thought it was meant to fix the screen corruptions on macOS.

However, good luck tomorrow and have a nice evening! 👍

ianpaschal, 2018-02-15:
@thopiekar Nope. I just realized what you meant now that I'm at work and paying more attention. I thought you were referring to something else also called PPA. Which I worked on a lot in the last month. Oops.

But in this case I have no idea how to fix it. Running from source seems to be fine as the correct commits are in both `Cura` and `Uranium`.

The gist of it was that the `PluginsModel` class took `parent` as an arg. I didn't see it being used so when I re-wrote the `PluginBrowser` I changed the model to take `view` as an argument which would do filtering (to try and remove logic from QML).

This then broke the General page of preferences where two items from the plugin model, sending anonymous slice data and checking for upgrades, also used it, and supplied a `parent` argument which was now no longer used.

The quick fix was simply to return `parent` to being the first argument and instead make `view` a second argument with a default value of `installed`. And of course in the plugin browser it's now created as: `PluginsModel(None, "available")`. It would appear your copy of the plugin browser is still attempting to do `PluginsModel("available")`, hence `parent` has wrong type: `str`.

So it's Cura, not Uranium, which is not up to date in the PPA. I think that's something you'll need to check for yourself.

thopiekar, 2018-02-15:
Yep, the other secret thing you worked on 😄 
Hmm, I will observe it. Currently I'm on Windows at the moment, so can't check again now..

thopiekar, 2018-02-15:
Works now like a charm. However, the coloring issue is still there, but that's another issue :)

closed 2018-02-15 by thopiekar

ianpaschal, 2018-02-16:
Hmm right. That's a fairly easy fix. Maybe I will just do it this afternoon if I have a minute. I wasn't really familiar with UM.Theme when I wrote the plugin browser code. 😅

thopiekar, 2018-02-16:
Well, there is also more in the "Printer" tab. I'm probably the worst case user, because I use Ultimaker white theme on a black desktop :sweat_smile:

ianpaschal, 2018-02-16:
...oh dear. Well in that case we'll see. In the next research sprint I am thinking to apply my Node/UI skills to try and rewrite a lot of the theme handling to use more variables and more reusable parts. It may not be worth trying to update everything before that. Would be great if you could document these cases of theme weirdness.