Ultimaker/Cura #3304: CuraEngine.exe has stopped working - When enabling Ironing feature

easycheese, 2018-02-13:
Slicing while using the ironing feature fails for quality > .25

**Application Version**
3.1.0

**Platform**
Windows 10

**Qt**
Website version

**PyQt**
Website version

**Display Driver**
DxDiag attached

**Steps to Reproduce**
Load Prusa i3 Mk2 profile
Load any STL
Set quality to .4mm
Check 'Enable Ironing' 

**Actual Results**

CuraEngine.exe has stopped working

**Expected results**
STL slices successfully

**Additional Information**
If I manually set the layer height, I need to set it to .25 or lower before it will successfully slice

[cura.log](https://github.com/Ultimaker/Cura/files/1721997/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/1721995/DxDiag.txt)
[PI3MK2_Laundry Holder.zip](https://github.com/Ultimaker/Cura/files/1722006/PI3MK2_Laundry.Holder.zip)

diegopradogesto, 2018-02-14:
I could reproduce it in 3.1, but trying to do the same in 3.2 it's not happening. So I encourage you to try the new version and test whether the bug is still there or it is fixed.

easycheese, 2018-02-22:
Ah yep, didn't realize that 3.2 was out.  It's working there, thanks!

closed 2018-02-22 by easycheese