Ultimaker/Cura #3306: Incorrect TOP Layers for parts of the model

xcorwin, 2018-02-14:
**Application Version**
3.2.0
**Platform**
OSX

**Steps to Reproduce**
SLICE both the models in the ZIP file.

Im using 1.2mm walls, 3 loops, slicing at .2mm resolution.

You will see the Shackle (seperately) gets the correct TOP layers (6 layers as per my settings)

[T.zip](https://github.com/Ultimaker/Cura/files/1723042/T.zip)

But the same shackle inside the model, gets only 3 layers of TOP (ceiling) and then 3 additional layers of INFILL!

Ive tried rotating the model to see if it may trick the slicer into doing it differently, but to no avail.

**Actual Results**

As can be seen from these pictures:
[image]
[image]
The stand-alone shackle on the left has all 6 top layers correct, but as part of the bigger model, it prints infill on the top 3 layers of the roof.

**Expected results**
I expect the slicing of the shackle to be consistent regardless where it is contained in the model

smartavionics, 2018-02-14:
Set the Slicing Tolerance experimental setting to "Exclusive" and it looks better so this is probably a feature rather than a bug. I do think that the slicing tolerance setting is rather hard to understand and I nearly always end up trying the different options and comparing the results.

xcorwin, 2018-02-15:
Hi

Enabling ‘Slicing Tolerance’ to ‘Exclusive’ did improve the slicing result. It is now consistent with the stand-alone and with other slicing programs
Thanks.

--Cor

xcorwin, 2018-02-15:
Wall thickness 1.2 and top/bottom 0.8 it works...

Wall thickness 1.2 and top/bottom 1.2 it does not work...

It looks to me as if internal roof cannot have more than 4 layers... but instead of slicing it for the TOP 4 layers, it slices for the first 4 out of the number it is supposed to slice... even with the tolerance set to Exclusive.

smartavionics, 2018-02-15:
> It looks to me as if internal roof cannot have more than 4 layers... but instead of slicing it for the TOP 4 layers, it slices for the first 4 out of the number it is supposed to slice... even with the tolerance set to Exclusive. 

That's interesting, I've not come across that before.

BTW, I did import that model into FreeCAD and it reported all sorts of errors so it could be that the model has some problems that are fooling Cura?

xcorwin, 2018-02-15:
From: Mark Burton <notifications@github.com>
Reply-To: Ultimaker/Cura <reply@reply.github.com>
Date: Thursday, 15 February 2018 at 09:22
To: Ultimaker/Cura <Cura@noreply.github.com>
Cc: Cor Winckler <cor.winckler@pbtgroup.co.za>, Author <author@noreply.github.com>
Subject: Re: [Ultimaker/Cura] Incorrect TOP Layers for parts of the model (#3306)

>BTW, I did import that model into FreeCAD and it reported all sorts of errors so it could be that the model has some problems that are fooling Cura?
That is entirely possible, but the funny part still is that the little Shackle on it’s own does not (ever) exhibit the behaviour... only when sliced as part of a bigger model. Ive not had time to try and re-create the smallest possible set of constructs that would behave like this...

For now, ive simply sliced this particular model with more support and only 4 layers of top/bottom, to try and avoid the problem. I would prefer more layers...

--Cor

Ghostkeeper, 2018-03-08:
I've not been able to find anything wrong with either of the models. The normal vectors are a bit off on the separate shackle but that's not used by Cura. There don't seem to be missing or double walls anywhere. Even OpenSCAD claims it's watertight and manifold.

The culprit seems to have something to do with Ignore Small Z Gaps. If you disable that setting it slices normally. Makes sense too, because there is a small Z gap at that position. I wouldn't expect there to be skin below and above that infill then, though... I'm not quite sure what is going on.

diegopradogesto, 2019-09-10:
I can't see this issue happening in the latest stable (4.2) or the latest beta (4.3 beta).

Please feel free to open it it still happens to you.

closed 2019-09-10 by diegopradogesto