Ultimaker/Cura #3307: Release tags for 3.2.0 missing

onitake, 2018-02-14:
Cura 3.2.0 has been out for a while , but I don't see any release tags on the Github repositories.

Are you going to tag them, or did you change your release process?
If yes, what's the correct way to get the sources that correspond with a particular release?

alekseisasin, 2018-02-14:
We have separate branches, like 3.0,  3.1, 3.2. Where we store the release versions.

onitake, 2018-02-14:
Yes, I noticed. But these branches lack explicit release tags too.
Which commit is 3.2.0? Which commit would 3.2.1 be, if it is ever released?

Up until 3.1.0, releases were always tagged on Github. Was this policy changed in 3.2.0?

onitake, 2018-02-14:
FWIW, I noticed that @Ghostkeeper has just tagged 3.2.1, so I guess it was just a matter of time.

Thanks, closing this bug.

closed 2018-02-14 by onitake

Ghostkeeper, 2018-03-07:
Yeah we forgot about 3.2.0 and then I was thinking that I'd just tag 3.2.1. It's more useful to be able to reproduce 3.2.1 anyway.