Ultimaker/Cura #3308: Cura generates useless support

JulienDevillers, 2018-02-14:
**Application Version**
3.1

**Platform**
WIN7 SP1

**Qt**
 using the version from Ultimaker's website

**PyQt**
 using the version from Ultimaker's website

**Display Driver**

**Steps to Reproduce**
import this file : 
[CC_porte_tringle - 2.L.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1723134/CC_porte_tringle.-.2.L.curaproject.3mf.zip)

**Actual Results**
Cura generates useless support like here:

[image]

**Expected results**

**Additional Information**

ianpaschal, 2018-02-14:
Can you provide more information about how you're getting this to generate? Are you using PVA?

smartavionics, 2018-02-14:
It's related to the support towers setting. Turn off the towers and they should go away.

JulienDevillers, 2018-02-14:
@ianpaschal :
- To get this:  import the 3mf file. What other info do you need?
- What is PVA?

@smartavionics : you're right.

ChrisRiddell, 2018-02-14:
Yeah its related to support towers like @smartavionics said,

To be honest this has come up a number of times now here and on the forums maybe it should not be on by default.

ianpaschal, 2018-02-14:
@JulienDevillers Yes, it's the support towers. PVA is a kind of filament.

@ChrisRiddell I'm not a big fan of them either but I believe they're on by default because the materials department at Ultimaker says they produce better prints when using PVA, which is why I asked @JulienDevillers if he was using PVA which would change this from unexpected to expected behavior.

closed 2018-02-14 by ianpaschal