Ultimaker/Cura #3309: Cura 3.2 always crashing while slicing : CURAengine.exe stop working

nurayha, 2018-02-14:
Cura 3.2 keep on crashing while slicing stl file.

smartavionics, 2018-02-14:
With all due respect, that's a completely useless report. You need to provide more info such as what profile you were using, what model, etc. Without that info, there's no chance that your issue will be investigated. Thanks.

nurayha, 2018-02-14:
Same problem with another stl file.
[image]

smartavionics, 2018-02-14:
OK, thanks for providing some info. Now, what version OS are you using? How much memory is in the machine? Can you locate the cura.log file and post it here?

Actually, I am not a good person to talk to about this because I don't use Windows. Hopefully, someone else can pick this up.

nurayha, 2018-02-14:
This is my computer spec
[image]

I'm sorry, but may I know what is the cura.log file?

Vandrasc, 2018-02-14:
Hi @nurayha, you can find your log files in Cura itself, Launch Cura -> Help menu->Show configuration folder or in C:\Users\<username>\Appdata\Roaming\cura\3.2

I doubt the log will show something about the crash though, I do hope so. 

This happens after the upgrade from a previous version? Could you check your **Roaming and Local** folder and see if there is another version mentioned, except the 3.2? If so, then you can remove the older directory and restart Cura, be sure to backup your Cura folder if you don't want to lose any important files. 

Please let us know if this fixed your problem, thank you!

nurayha, 2018-02-14:
Hi @Vandrasc ,
Thanks for the advice. I will try it and update you guys soon during office hour in my country, since I installed the Cura at the office.
Cheers.

nurayha, 2018-02-15:
Good day everybody,
So I guess this is where the log data is located.
So what should I do next?
Now even with the Cura 3.2.1 version,  I'm still having the same problem. I think it could be my computer problem since it works ok in my other PC.
[image]

EvilGreenOne, 2018-02-16:
I am having this same issue today after updating to 3.2.1 from 3.1.

Reinstalled 3.1 in parallel with 3.2.1; 3.1 slices the same file with the same profile fine.

All previous profile directories have been removed from /AppData/Roaming/cura

System Info:
Edition: Windows 10 Pro
Version: 1709
OS Build: 16299.248
System Type: 64-bit
Installed RAM: 32 GB

```
Windows Error Event 1000 for the process:
Faulting application name: CuraEngine.exe, version: 0.0.0.0, time stamp: 0x5a840e96
Faulting module name: CuraEngine.exe, version: 0.0.0.0, time stamp: 0x5a840e96
Exception code: 0x40000015
Fault offset: 0x000000000020e212
Faulting process id: 0x3a20
Faulting application start time: 0x01d3a76391e6b56c
Faulting application path: C:\Program Files\Ultimaker Cura 3.2\CuraEngine.exe
Faulting module path: C:\Program Files\Ultimaker Cura 3.2\CuraEngine.exe
Report Id: 20c9f2d0-7701-4061-8529-27ec11681f95
Faulting package full name: 
Faulting package-relative application ID: 
```
Cura Log file with this event only - Startup to Crash.
[cura.log](https://github.com/Ultimaker/Cura/files/1732737/cura.log)

File sliced does not matter, but for this test I was using a TPU profile and this model:
https://www.thingiverse.com/thing:2510970

Vandrasc, 2018-02-17:
Thank you both for sharing, unfortunately we are not able to reproduce, could you please zip your 3.2 user configuration and share the folder with us? 
And you both removed your older configuration (3.0, 3.1) and it still occurs with a clean install?

EvilGreenOne, 2018-02-17:
This was an upgrade install.  However, prior to adding to this report I followed the trouble shooting tips given previously and had removed all of the previous user directories for prior versions.

If it will help, I can remove both 3.1 (Which I reinstalled to get some slicing done), and 3.2 and try with a completely blank config.  That is not really a solution for me though as I have a customer printer configured and my own library of materials and profiles.

Attached is the entire 3.2 user directory from the affected install.
[3.2.zip](https://github.com/Ultimaker/Cura/files/1733748/3.2.zip)

I did that totally clean install with no previous profiles to be auto-imported and the cura engine does not crash.  I forgot just how many of the useful features are hidden by default.  It's like using a completely different application!  I guess it is back to 3.1 for now just because I do not have the patience to re-setup my printer and profiles.

fieldOfView, 2018-02-17:
I have not looked at your files, but check the layer height and the initial layer height. I have seen someone reporting that the crash happens when the layerheight is higher than the initial layer height.

EvilGreenOne, 2018-02-17:
With the user profile restored that makes it crash.  I tried the following:
Layer Height: .2
Initial Layer Height: .3
- Crash

Layer Height: .2
Initial Layer Height: .2
-Crash

Layer Height: .4
Initial Layer Height: .3
-Crash

None of the combinations I tried, Lower/Higher, Even, Higher/Lower remediated the issue.

fieldOfView, 2018-02-17:
Thanks for trying

Ceteak, 2018-02-18:
See if adaptive layer is on , mine kept crashing when slicing as well, but would only crash the slicing part of the program i could still move the STL and change settings when i turned adaptive layer off it sliced fine

Vandrasc, 2018-02-18:
@EvilGreenOne, I could indeed reproduce your issue with your configuration, I think it has to do with the Enable retraction children settings, Retraction speed, Retract speed and Prime speed. I noticed that the retraction was disabled and after enabling it those children settings show an error value. I customized the children settings (from 0.0 change to 0.1mm/s) of the retraction and I was able to successfully trigger a slice. So, I would say that the fault is in the profile? As workaround try to change the retraction speed, retract speed and prime speed to an acceptable value and then disable the retraction. We will have a further look but for now, I hope this workaround works for you.

EvilGreenOne, 2018-02-18:
@Vandrasc That indeed did the trick.  Thank you for the support.

I think that the way this came about is that the default values when you "create" a new material fills those two values with 0's.  If I am working with TPU or TPEE, I pretty much run with no retraction so I didn't even think to change those to some other value as Retraction was disabled in my Flexible profile I usually start from.  In previous versions if Retraction was off, I guess it just properly ignored those settings?

I am glad we found a resolution to the issue.  Now I can go take the new features for a spin.

nurayha, 2018-02-19:
Hi everyone,
Thanks for all the advice. I'm glad some of you already solved the issue, but the issue with mine haven't resolved yet. My Cura 3.2.1 still keep on crashing while it is halfway slicing an item. I have checked and tried all method suggested by you guys, but it still happens again.
This is my cura log for your reference.
[cura.log](https://github.com/Ultimaker/Cura/files/1735701/cura.log)

Thank you.

Vandrasc, 2018-02-19:
Hi @nurayha, if it works ok on your other pc it may be that there are some configurations which are causing the crash, could you upload your user configuration as well (the folder where the log is located)? 

@EvilGreenOne anytime, glad it did the trick as well. Yes, it may be, we will try to investigate it and make a fix for it. 
And, thank you for working with us :)

nurayha, 2018-02-19:
Hi @Vandrasc 
Here's the cura configuration as requested.
[cura.zip](https://github.com/Ultimaker/Cura/files/1735879/cura.zip)
Hopefully you can find something that causes the issue in here.
Thank you.

Vandrasc, 2018-02-19:
Is it possible to attach the entire 3.2 folder? The one containing the quality, machine definitions, extruders? 
Thank you for your quick reply

nurayha, 2018-02-19:
Hi @Vandrasc,
Here you go.
[3.2.zip](https://github.com/Ultimaker/Cura/files/1735951/3.2.zip)
Thank you.

Vandrasc, 2018-02-19:
Oke, @nurayha easily reproduced. The slicing engine stops working when the default temperature is set to 0, though it displays warning values it should keep slicing. Any possibility to print with a higher temperature than 0 degrees? I set the default temperature to 0.1 and it successfully triggered a slice. 
I will make a bug report and will be fixed in the next release. 
Good catch, both of you!

nurayha, 2018-02-19:
Hi @Vandrasc,
Thank you for your prompt response.
Yup, it turns out that the value 0 in Default Printing Temperature causes the crashing. It wasn't an issue when I did the same with Cura 3.1.
Thank you so much for helping us solve the problem. Appreciate it.
Cheers! :)

Vandrasc, 2018-02-19:
Devs CURA-4982/4983

Vandrasc, 2018-03-09:
Both issues fixed. Thank you all for reporting it!

closed 2018-03-09 by Vandrasc