Ultimaker/Cura #3310: Cura 3.2 crashes at slicing

filledk, 2018-02-14:
Hello there, recently i bought a Ultimaker 3,

but my cura crash when i try to slice a STL file, when i click "prepare" (i have manual slicing) near the slicing it just closes cura,

this is really a shame as i bought the Ultimaker 3, i currently use cura 2.7 as that works fine...

sorry but im a newb to post here on github, hope u understand,

ps i cannot seem to find any crashlog ;-S

diegopradogesto, 2018-02-14:
When you create a new issue there is a template that you can fill out with useful information for us. Could you please complete this information so we can help you better?

alekseisasin, 2018-03-07:
Hi @filledk, Do you still have same issue with latest Cura 3.2.1 version?

closed 2018-03-09 by ianpaschal

Ghostkeeper, 2018-04-04:
You'll be able to find a log file when you start Cura and go into Help -> Show Configuration Folder. There should be a cura.log there somewhere. Or if that fails, see the instructions here: https://github.com/Ultimaker/Cura#logging-issues