Ultimaker/Cura #3311: Extruder temperature 0°C

sfornengo, 2018-02-14:
**Application Version**
Cura 3.2 / 3.1

**Platform**
Ubuntu 16.04

I am using Cura AppImage. My printer is a printrBot simple metal.
The reported extruder is 0°C and the setpoint is also 0°C... the printing never start.
I must close and restart cura 3.2 multiple times and sometimes the setpoint and the extruder temperature are ok, and then I can succeed to print.
However, during the nozzle temperature rise, the displayed temperature is not updated although I can confirm it is heating. This problem of update was also present in Cura 3.1.

alekseisasin, 2018-02-14:
Hi @sfornengo  . What do you mean under "succeed to print"? Do you mean  print over network or save as file ? Did you have same problem with Cura 3.0?
Did you try contact Printbot and ensure that this happens with all printers (printrBot simple) of this brand. 
We do not have an issue with Ultimaker printers and it is difficult to find the reason why it does not work without having the Printbot printer.

sfornengo, 2018-02-14:
I am printing via USB.
It worked with old Cura version, so something has changed in 3.1 and 3.2

Ghostkeeper, 2018-03-07:
Perhaps you could load up Prontrface and see what the printer responds when you send the "M105" g-code command? That's what Cura uses to request the temperature.

Raul-VLC, 2018-03-20:
Hi, I have the same problem. 
Extruder temp 0º and I check with PronterFace and says 16º of extruder temp.

[image]
[image]

So, in Cura 3.2.1 it doesn't start to print, always stay in "printing 0%". 
In older Cura, we printed normally, with the same machine yesterday.
I cancel the print, and I can move correctly X, Z & Y axis... 
Can you help me please?

closed 2024-12-10 by github-actions[bot]