Ultimaker/Cura #3315: Cura 3.2.1 Build plate model reloaded and centered when reloading all models

PvtPrivate, 2018-02-14:
**Application Version**
Cura 3.2
Cura 3.2.1

**Platform**
Windows 10

**Steps to Reproduce**
1. Import any model
2. Reload all models (F5)

**Actual Results**
All models are reloaded, but the buildplate model and nozzle model also gets reloaded and centered.

**Expected results**
All models are reloaded, no changes on build plate and nozzle model.

**Additional Information**
Before:
[image]
After:
[image]
[image]

Vandrasc, 2018-02-15:
Hi @PvtPrivate, good catch, thank you for reporting it!

Devs see Cura-4963!

ianpaschal, 2018-02-15:
Fixed as of today. Should be in the next release.

closed 2018-02-15 by ianpaschal