Ultimaker/Cura #3316: Changes in Cura profile not reflected after saving

Jfactor91, 2018-02-14:
Cura 2.7 because I can't import my profiles into anything newer. I've updated a profile with my preferred settings, saved it. When I move to extruder 2, select that same saved profile, completely different numbers show up for almost all of the settings. I'm going through an entering in numbers manually, this is a waste of time. 

Jordan

alekseisasin, 2018-02-15:
@Jfactor91 We had issues related with multiple extrudes and after moving from one version to another we apply version upgrade to update user-data settings, profiles and in most cases it worked. Probably some of settings are cannot be migrated to a new version and it is a reason why you cannot update.

I am not sure if we can provide a very fast fix for your profiles, from 2.7 version were made a lot of changes especially for multiple extruders.
My proposal to install a new version 3.2.1. It is much faster and does not have an issue with multiple extruders and manually update profiles, sorry for this if it will cost a lot of time for you.

Ghostkeeper, 2018-03-08:
Keep in mind that extruder 1 and extruder 2 will almost always have completely different numbers. Unless there is a little shackle near the setting, the setting value that you enter will be for only the selected extruder. Saved profiles also save different settings for each extruder separately, so if you saved your profile while extruder 1 was active, it would also save extruder 2 in the same file. If you then load that profile it'll load extruder 2 again. You can't use the profile import/export to move settings from one extruder to another.

You can "copy value to all extruders" setting-per-setting by right-clicking on the setting and selecting that option in the context menu.

closed 2018-03-28 by ianpaschal

Ghostkeeper, 2018-04-25:
There is now also a "Copy all changed values to all extruders" option that makes this easier to use.