Ultimaker/Cura #3317: Incomplete files for 3.2

hairbandrocker76, 2018-02-14:
Since the latest update, I'm not getting full prints.  The print will just stop early as if it is complete or it will get a SD read card error and sit there cooking the filament in the nozzle.  Has happened twice over 5 prints.  I make sure the file is completely copied to the sd card before ejecting it but it still happens.  Please look into this.  Thank you.
-->

**Application Version**

**Platform**

**Steps to Reproduce**

ianpaschal, 2018-02-15:
Can you share some example files that did not print successfully (mesh and gcode)? That way we can analyze them for potential problems but right now it could be a hundred things, many of which are not related to the software.

hairbandrocker76, 2018-02-15:
I'm sorry, I don't have any now.  I deleted them and re did the prints.
Invite dies it again I'll hang on to it and send it out to you.  Thanks

ianpaschal, 2018-02-15:
That's understandable. Just deleting and trying again is my first instinct as well in such cases.

Keep us posted.

Ghostkeeper, 2018-03-08:
Were you using adaptive layer height with these prints?

hairbandrocker76, 2018-03-08:
Hello.  No I wasn't.  I haven't used that option yet.  I haven't come
across this problem again thankfully.  I'll let you know if it does
though.  thank you for the follow up.

Thomas

Ghostkeeper, 2018-04-03:
We aren't able to reproduce this.

I think Hairbandrocker76 hasn't been able to reproduce this either, so let's hope it fixed itself in the meanwhile.

closed 2018-04-03 by Ghostkeeper