Ultimaker/Cura #3318: 3.2.0 final - Multiple objects on build plate slice only to lowest when using Adaptive Layer height

desrtfx, 2018-02-14:
**Application Version**
3.2.0 final

**Platform**
Windows 10x64, Windows 7x64 (tested on two different installs on two machines)

**Display Driver**
Nvidia Graphics cards on both machines with latest drivers - one GT720M (W10 machine), QuadroFx (W7 machine)

**Steps to Reproduce**

+ Load multiple models in the build room, (tested with multiple models, for example: https://www.thingiverse.com/thing:2443261) 
+ turn adaptive layer height on, 
+ slice, 
+ switch to layer height preview

**Actual Results**

Slicer slices only as far as the height of the lowest object is. Remaining height of the taller object is ignored and not sliced.

See: https://i.imgur.com/y8OCZfi.png

**Expected results**

Full slicing as here: https://i.imgur.com/qvIVDTq.png

**Additional Information**

This problem only appears when slicing multiple objects. Single objects work without problems even though the surface quality is inferior to "normal" - non adaptive layer height prints.

The order in which the models are imported into Cura doesn't matter. Smallest always wins.

Grouping the models also has no effect. 

Changing parameters for adaptive layer height also has no effect on the listed problem.

diegopradogesto, 2018-02-15:
We already noticed that. Thank you!
Devs, it is related with CURA-4869

Ghostkeeper, 2018-03-03:
This was fixed for the 3.3 release, mainly by this commit: https://github.com/Ultimaker/CuraEngine/commit/ea0642ee3dbbf8e5e88a255c7b5d2c7fd03873b5

closed 2018-03-03 by Ghostkeeper