Ultimaker/Cura #3321: [discussion] QtQuick Controls 2

fieldOfView, 2018-02-15:
Cura 3.2 switched the sidebar to QtQuick Controls 2, and other parts of Cura are now beginning to adopt it too (eg the Plugin browser). Before this trend continues, I would like to highlight and discuss some pretty major drawbacks of QtQuick Controls 2

* No more "system" styling
Cura (and Uranium) live by the convention that any dialog that pops over the main window should be "system-styled", in order to maintain consistency with system-supplied dialogs such as file open/save dialogs and simple yes/no/cancel dialogs. Uranium applications need to be able to live without the Theme layer. It has always been possible for QML controls to use system theming, but this functionality it dropped from QtQuick Controls 2. This means that all of Uranium is going to have to adopt a non-system theme/style, unless we maintain a mix of QtQuick Controls 1 and 2.

* No more "style" property
The style property is used throughout Cura to have controls that look and feel in a consistent way. Inlining all this style and behavior is going to lead to a lot of code duplication. I don't know if creating a set of subclassed controls is going to be a viable solution to this issue.

* Broken text rendering on some platforms
The QtQuick Controls 2 Label no longer uses Native text rendering by default, leading to broken text rendering on some platforms (notably OS X). This could be fixed by introducing a subclassed UM.Label control to replace the Label control (but it would still suck to have to do this).

All in all, I would vote against further adoption of QtQuick Controls 2, unless well-tested solutions to these issues are found.

ianpaschal, 2018-02-15:
> Cura (and Uranium) live by the convention that any dialog that pops over the main window should be "system-styled", in order to maintain consistency with system-supplied dialogs such as file [...]

Do they? Themes are a built-in part of Uranium, not Cura. Seems weird to develop a GUI application and not touch the appearance of the UI. Also Qt has some default styles which could be fallen back (?) to.

> No more "style" property
To be honest, I see this as a big plus. If all of a component's style was kept in the style component, sure, nice way to delineate it from event handlers and other such stuff, but it's not.

Obviously I see the problem of how this would wreck themes but @LukeChen-Ultimaker and I both feel the current theme approach is borderline impossible to use. Defining reusable components as one would do in, say, React or Vue is a better approach, IMO, with only things such as color and font size taken from aprox. 10 theme variables.

> Broken text rendering on some platforms
Why not just include it in the aforementioned reusable components? ;)

ChrisTerBeke, 2018-02-15:
Given 1.x will likely drop support in the future, updating to a newer version is always good. However it should be done in the right way, and as @ianpaschal indicates that means rewriting quite a bit of the QML code into reusable components. I think that process will take a while and could happen on a separate branch while keeping the mainline code so we don't break user experience for the next few releases.

fieldOfView, 2018-02-16:
>> Cura (and Uranium) live by the convention that any dialog that pops over the main window should be "system-styled", 
> Do they? Themes are a built-in part of Uranium, not Cura. 

Yes, they do. Even though Uranium defines the Theme system, none of the QML files in the Uranium repository use it. The "client" application defines a theme to use, Uranium has no theme of its own and relies on system-styling.

Come to think of it, the move/rotate/scale panels may be the only exception here, but @awhiemstra will agree that that's not the way it is supposed to be either (Lay Flat has no business being in Uranium). (PS: yes, that's another thing *I* put in there that should have lived in Cura instead)

ianpaschal, 2018-02-16:
> Yes, they do. Even though Uranium defines the Theme system, none of the QML files in the Uranium repository use it. The "client" application defines a theme to use, Uranium has no theme of its own and relies on system-styling.

The "do they" was meant primarily about Uranium applications functioning with no theme.

Everything needs some source of default styling and while it may be that right now Uranium takes its default styles from system and can be overridden I don't see any reason why it wouldn't use one of the built in Controls 2.x styles instead.

nallath, 2018-02-20:
The main reason why i switched to qtquick controls for the settings was speed; The controls (and there are a *lot* of them) simply load much faster. 
I do agree that unless we find good solutions for these issues, that we should not continue to include more of these. The whole styling (eg; how to we facilitate "native" buttons) is a big issue and should be resolved first.

ChrisTerBeke, 2018-02-22:
According to https://github.com/qt/qtquickcontrols2, it ships with a native looking desktop style as well, but the default style is minimal. I guess it's possible to enable this native looking style if we want to.

ChrisTerBeke, 2018-02-22:
Found some examples on how to activate it: https://stackoverflow.com/a/48583334/3330513.

ianpaschal, 2018-02-23:
Yeah those default styles is what I was referring to here:

> I don't see any reason why it wouldn't use one of the built in Controls 2.x styles instead.

Seems like an acceptable fallback, especially since the existing system styling does not fool anyone about actually being native.

fieldOfView, 2018-02-23:
> especially since the existing system styling does not fool anyone about actually being native.

In the places where its done right it looks mighty convincing to me.

awhiemstra, 2018-02-28:
> According to https://github.com/qt/qtquickcontrols2, it ships with a native looking desktop style as well, but the default style is minimal. I guess it's possible to enable this native looking style if we want to.

Actually, that is not true. There is no real desktop style yet that actually uses the platform styling, apart from https://cgit.kde.org/qqc2-desktop-style.git/ but I do not know if that even works on other platforms than KDE. The "fusion" style specifically, is not actually native or even close to native.

There is also the additional issue that currently there is no real way to do "sub-theming" of specific parts of an application. You either pick a style for the entire application or you need to do everything through subclassing, which kinda goes against how styling is designed in the first place. Ideally, you would define a "Cura" QQC2 style and then be able to define whether something should use that or the platform style.

> Seems like an acceptable fallback, especially since the existing system styling does not fool anyone about actually being native.

Maybe on OSX. On both Linux and Windows, it looks fine in my opinion. Sure, things could be better, but saying "oh let's just switch everything to custom styling" is just throwing out the baby with the bathwater.

Some response to the original points as well:

> No more "system" styling

There is some development going on upstream in regards to this. QQC2 started life as a project to develop a set of controls that wouldn't completely choke an embedded system. Thus, the embedded/mobile use case had the highest priority, but they are now slowly adding new functionality relating to desktops.

> No more "style" property

I consider the way styling works in QQC2 an enormous improvement. Yes, you lose a bit of functionality in one place in favour of a great deal more in another. Additionally, it was simply needed. If you ever looked at a QQC1 control's implementation, you would have noticed the incredible amount of indirection happening there. For QQC2 they wanted much more lightweight controls, necessitating the removal of the style property.

> Broken text rendering on some platforms

This could easily be fixed by the aforementioned "cura" style. The style has an incredible amount of power over the individual controls, you can even add custom properties and everything. That said, there are some known issues with font rendering upstream that will hopefully eventually sort out all this.

fieldOfView, 2018-09-20:
I am closing this, as there's no use in keeping this open forever.

closed 2018-09-20 by fieldOfView