Ultimaker/Cura #3322: 2 color slicing overlapping filament

notmine642, 2018-02-15:
Hello, 

I am trying to print a QR code using 2 different color.
It seems that the slicer make some layer overlapping...
hard to explain, I have attached pictures, Git does not accept the project.
Thanks for your help
layer 10
[image]
layer 9

[image]

Ghostkeeper, 2018-03-08:
Github will accept your project or models if you put it in a .zip.

Ghostkeeper, 2018-03-08:
It's kind of hard to debug this without it occurring on my side. Maybe you can play a bit with the "Union Overlapping Volumes" setting, or "Alternate Mesh Removal"?

ianpaschal, 2018-03-27:
Haven't heard back from OP so I'm closing this for now.

closed 2018-03-27 by ianpaschal