Ultimaker/Cura #3323: Estimated Print Time is Wrong

SuperBoppy, 2018-02-15:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10 Home Version

**Additional Information**
The CURA display window print estimation time appears to be double of what the actual time the printer takes to print the object. For example, if CURA estimates a time of 3 hours, the print will only take 1 1/2 hours to complete. BTW, I didn't notice this error in previous versions - the time always seemed to be spot-on before.

ianpaschal, 2018-02-15:
Thanks for reporting.

Can you provide an example of the model and gcode so we can see what may be causing it?

Also, does this happen with _every_ model you print, or just some of them? I've been running 3.2.1 for a while and haven't noticed any big different in print time, especially not double/half. So reproducing might be a trick without the exact models/files in question.

SuperBoppy, 2018-02-16:
I have not printed too many things since upgrading, but it seems to be showing incorrect times on other objects as well. I printed an xyz calibration cube, and I remember CURA telling me that it was going to take around 45 minutes to print it. I wondered at that, because it only takes around 20 minutes to print the cube.
 
This print took 1h 56m. It took a little longer than my previous print, which took just under 1 1/2 hours, because I reduced the acceleration in the firmware from 2650 to 1000.  Still, the CURA estimate is an hour off from what the actual print time was. My speed settings in the firmware for this print are as follows:
 
Accel: 1000 (TEVO default is 2650)
Vxy-Jerk: 15
Vz-Jerk: +000.40
Ve-Jerk: 5
Vmax x: 225
Vmax y: 225
Vmaxz: 3
Amax x: 1000 (TEVO default is 2650)
Amax y: 1000 (TEVO default is 2650)
Amax z: 100
Amax e: 10000
A-retract: 3000
 
All values listed above are TEVO Tarantula defaults, unless otherwise noted.
 
I have attached a screenshot of CURA, the STL file, 3D Printer photos, and the gcode generated by CURA. 
 
Hope this helps.
 
Rick 
[Cura Troubleshoot.zip](https://github.com/Ultimaker/Cura/files/1732283/Cura.Troubleshoot.zip)

closed 2018-02-16 by SuperBoppy

Ghostkeeper, 2018-03-09:
Thanks for the data. I've added that data to the machine definition for the TEVO Tarantula. Hopefully time estimates for that printer will be better in the next release.