Ultimaker/Cura #3324: [Adaptive layer height] Wrong 'step size' may cause fractional motor steps value (it depends by leadscrew pitch, motor steps, etc)

noisemaker00, 2018-02-15:
Please add a validation to the `step size` into the Adaptive Layer Height feature because not all layer height can be achieved by the printer.
This value is strictly related to:

- lead screw pitch/diameter
- motor angle/steps

**If the layer height is wrong, the number of motor steps won't be an integer value.**
You can check it using Prusa Calculator layer height ([https://www.prusaprinters.org/calculator/](https://www.prusaprinters.org/calculator/)).

You can insert a section in the Printer Configuration to insert these information the perform a check when user insert the 'step size' suggesting the closest value.

eg. 
My printer configuration is:

- 400 steps/turn
- M8 lead screw with 1.25mm pitch

Let assume to use this configuration for Cura
- Set a 0.2mm base layer height
- Default step size (0.01mm) 

This cause wrong layer height, for example:
0.2 - 0.01 = 1.99mm (BAD, number of steps is not integer value)

According with the Prusa Calculator, on my printer configuration a 'step size' of 0.0125 is better, because:
0.2 - 0.0125 = 1.9875mm (OK)

smartavionics, 2018-02-15:
There is also the other problem of the layer height being inappropriate for a given printer due to the extruder performance limitations.

If the layer height is very small, the amounts of filament to be extruded for each line can be less than the resolution (steps/mm) of the extruder and depending on whether absolute or relative extrusion mode is in use can cause either blobs or missing filament to occur.

On the other hand, if the layer height is too large, the amount of filament to be extruded could exceed the extruders capacity and so that will cause under-extrusion/skipping/stripping, etc.

What is required is the means to be able to specify the min and max extrusion rates and then the layer heights should be constrained so that the extrusion rates commanded are within the allowed bounds.

ChrisRiddell, 2018-02-16:
I think I have a working example of this issue.

First curaprofile has 
Adaptive layer height enabled and in the photo the first marking I see gaps/under extrusion and the second marks as the head moves across it hits the part and has broken off one side before tried printing a number of times same results.
[image]

[UM2_Stand_Alt_1.curaproject.zip](https://github.com/Ultimaker/Cura/files/1730533/UM2_Stand_Alt_1.curaproject.zip)

Second curaprofile has 
Adaptive layer height Disabled, Layer height set to 0.25mm zero issues printed perfect.
[UM2_Stand_Alt_2.curaproject.zip](https://github.com/Ultimaker/Cura/files/1730544/UM2_Stand_Alt_2.curaproject.zip)

Ghostkeeper, 2018-03-08:
How would this work with delta printers?

smartavionics, 2018-03-08:
I can't give a definitive answer but I have 2 delta printers and I have never had any issues like the above that are related to layer thickness.

What is the problem of the number of steps not being an integer value? The firmware will just use the nearest integer, don't you think? If that leads to an error of +/- 1 step, what does that equate to in terms of x/y distance?

Ghostkeeper, 2018-04-03:
Thinking about it again, I guess delta printers are out of the question anyway because even normal horizontal layers are not on one Z step. There is no Z step in fact.

> What is the problem of the number of steps not being an integer value? The firmware will just use the nearest integer, don't you think? If that leads to an error of +/- 1 step, what does that equate to in terms of Z distance?

The problem of this is that on some layers it will round up and on other layers it will round down. This means that some layers are higher than others. Also, the extrusion indicated by Cura will be oblivious to this rounding so you'll get over- or underextrusion.

andrew-davie, 2018-05-09:
Stepper motors (at least some) also do microstepping, so the "step size" is usually much finer. On the Creality machines, with a stepper resolution of 1.8° and 16 microsteps, the actual Z resolution is 0.04 mm for whole-steps, but if you consider micro-stepping then it's 0.0025 mm (although with some inaccuracy due to the way microstepping works). The point I'm making is that the OP does not appear to be considering microstepping as also being important when discussing adaptive layer height. IMHO.

Ghostkeeper, 2018-06-08:
For some printers there is apparently the concern that the motor is not strong enough to hold the build plate in a certain microstep position for a longer period of time.